littlefs-project/littlefs #419: Am I running out of space?

kazola, 2020-04-29:
Hi,

I am unable to understand a couple things about latest littlefs after running some example program creating 100 files. I am afraid I may be doing something quite stupid or maybe I'm at a bottleneck.

My NAND memory = 1024 erasable blocks, size 128 KB. Pages = 2048 bytes long, so 64 pages / block. Next comes the function I use in between littlefs and my memory driver. The one for writing is analogous. The low-level driver works fine when tested standalone.

```
int user_provided_block_device_read(const struct lfs_config *c,
                                    lfs_block_t block,
                                    lfs_off_t off,
                                    void *buffer,
                                    lfs_size_t size)
{
    uint16_t p;

    // A) starts slowing at 40, 15 seconds
    //p = (block * MEM_PAGES_PER_BLOCK);
    //p += (off / MEM_BYTES_PER_PAGE);

    // B) stops at 77 quite fast
    p = (block * MEM_PAGES_PER_BLOCK);
    p += off;

    hook_read_nand(p, 0, (uint8_t*)buffer, size);
    return 0;
}
```

As you can see, I have tried 2 implementations. The first one A) allows me to create all 100 files in my test, although around file 40 it starts taking 15 seconds and more! Normally I can create the file in around 1 second. The second one B) is faster, although at file 77 LFS says "superblock has become unwritable". Let me share my configuration next.

```
#define MEM_BYTES_PER_PAGE       (2048)
#define MEM_PAGES_PER_BLOCK    (64)
#define MEM_BLOCK_SIZE                  (2048 * 64)
#define MEM_BLOCK_COUNT             (1024)
#define LFS_READ_SIZE                     MEM_BYTES_PER_PAGE
#define LFS_PROG_SIZE                    MEM_BYTES_PER_PAGE
#define LFS_LOOK_SIZE                     (MEM_BLOCK_COUNT / 8)
#define LFS_CACHE_SIZE                  MEM_BYTES_PER_PAGE
#define LFS_BLOCK_SIZE                  MEM_BLOCK_SIZE
#define LFS_BLOCK_COUNT              MEM_BLOCK_COUNT

const struct lfs_config lfs_cfg =
{
    .read  = user_provided_block_device_read,
    .prog  = user_provided_block_device_prog,
    .erase = user_provided_block_device_erase,
    .sync = user_provided_block_device_sync,

    // whenever these values change, reformat memory
    .read_size = LFS_READ_SIZE,
    .prog_size = LFS_PROG_SIZE,
    .block_size = MEM_BLOCK_SIZE,
    .block_count = MEM_BLOCK_COUNT,
    .cache_size = LFS_CACHE_SIZE,
    .lookahead_size = LFS_LOOK_SIZE,
    .block_cycles = 500
};
```

2048 bytes is quite a big number but since it is the page size in the memory, I think is correct. I am afraid the function may be wrong (although it works for some files) and / or this provokes superblock getting out of space to write its metadata.

Thanks in advance.

e107steved, 2020-05-01:
Does your low-level routine expect a page number, or a byte offset?
Assuming a byte offset, shouldn't you have:
`p = (block * MEM_BLOCK_SIZE) + off;`

Also, unless required by the device, LFS_READ_SIZE can be quite small - littleFS sometimes only needs 16 bytes or so, and will read multiples of this if it needs more

kazola, 2020-05-01:
Hi @e107steved ,

thanks for your input :)

The low-level routine expects a page number.

I think you are totally right about LFS_READ_SIZE. However, this would only make my program slightly slower, but not provoke the problem / issue I am seeing, am I right?

e107steved, 2020-05-01:
Then you need to take the offset into account as well when calculating p, like you did in your first calculation. That looks correct to me.
If you reduce the page sizes, you'll also need to pass the 'residual' offset within the page to your routine; at a guess:
 
```
    p   = (off / MEM_BYTES_PER_PAGE);           // Page offset within block
    q = off - (p * MEM_BYTES_PER_PAGE);       // Offset within page
    p += (block * MEM_PAGES_PER_BLOCK);
    hook_read_nand(p, q, (uint8_t*)buffer, size);
```
Maybe you have two separate problems.

As you say, reducing the read/write block size will only affect speed - it was just a passing comment.

kazola, 2020-05-01:
Thanks Steve :)

I use this page size because I have enough RAM and the minimum writable unit in this NAND memory is 2048 bytes. Please let me know if I am misunderstanding anything here since my guess is switching to a more granular NOR will allow me to do smaller chunks but I cannot with the current memory.

Let's see if someone has an input about Littlefs slowering. I have seen it in other posts. My guess is NAND writing so big chunks make the superblock full too fast, but I may be saying something stupid because I do not know littlefs internals that well.

thrasher8390, 2020-05-08:
Where are you writing the new files (directory path)? As you get more and more files in a single directory LFS will create additional Linked Metadata Pairs. The slow down you're seeing may be due to a traversal that needs to occur before each new file is created. 

Once you get to file 1000, the file system may be traversing through up to 10 metadata pairs before it finds space to add the new file. I would recommend trying to limit the # of files you place in a single directory.

closed 2021-11-30 by kazola