littlefs-project/littlefs #420: Restrict the read size

keck-in-space, 2020-05-01:
Hello,

Is it possible to restrict the read size?

For example, our chip allows reads of any size, but our internal static buffer is only 256 bytes.

```c
#define LFS_CACHE_SIZE    256U
uint8_t readBuffer[LFS_CACHE_SIZE];
uint8_t writeBuffer[LFS_CACHE_SIZE];
uint8_t lookBuffer[1024] __attribute__((aligned(32)));

struct lfs_config cfg = {
    // block device operations
    .read = &lfs_read,
    .prog = &lfs_write,
    .erase = &lfs_erase,
    .sync = &lfs_sync,

    // block device configuration
    .read_size = LFS_CACHE_SIZE,
    .prog_size = LFS_CACHE_SIZE,
    .block_size = 4096,
    .block_count = 8192,
    .block_cycles = 500,
    .cache_size = LFS_CACHE_SIZE,
    .lookahead_size = 1024,

    .read_buffer = readBuffer,
    .prog_buffer = writeBuffer,
    .lookahead_buffer = lookBuffer,
};
```

The problem arises when a read request is sent to LFS that is larger than 256. Should those requests be chopped up into multiple requests at a higher level than LFS? Or should LFS be aware that the static buffer is only of size 256?

In the configuration shown above, I'm seeing read requests of up to 4096, the block size.

This is in LFS v2.2.1.

thrasher8390, 2020-05-04:
What is the concern of LFS requesting a read of 4096 bytes if your external chip can support it? Are you worried that you're taking up to much processing time? (Interested in the response to this since I'm currently considering cutting over to v2.2.1)

I never actually noticed if the reads requested to my block device drivers was `<=` my read cache size but I would expect LFS to be reading into the 4096 byte buffer that was passed to it. The cache size in my mind should only limit the amount of meta data that LFS reads into memory to perform it's own internal operations.

I'll be interested to see what others have to say about this though.

keck-in-space, 2020-05-05:
Hello! The issue is that I'm using static allocation and my read buffer is only of size 256 bytes. I do have some leeway to increase this, but the real issue is that LFS appears to be requesting a read that it has no place to put the result of. Since I allocate the buffers and no dynamic allocation is allowed since I've defined the ```#DEFINE LFS_NO_MALLOC``` flag.

I would think LFS should limit the max read size to the given size of the read_buffer, otherwise it won't fit in the given buffer.

kazola, 2020-05-05:
Hi @keck-in-space ,

I hope I am understanding you correctly. The input parameter in the .read function can point to a larger buffer. Then, you read inside your .read function and put your result in such pointer. If the input size parameter is > 256, just read twice, and so on.

1. Can you post your .prog, .read. and .erase functions? 
2. Also, have you tried, just for the sake of eliminating options, what happens if you allow dynamic allocation?

keck-in-space, 2020-05-05:
Hello, @kazola! That's what I'm asking, basically. Should I be the one chopping up these reads now, when LFS v2.1.4 apparently did that itself before.

1. Sorry, I can't really share any part of this project. The implementation of those functions are very simple, and really the only relevant part is that LFS requests a read of size 4096 when my read_buffer size is capped at 256.
2. For this particular projects dynamic allocation is out of the question, so I haven't tried that. It could be useful for LFS debugging purposes, but for now, I've rolled back to 2.1.4 and haven't had any read size issues.

kazola, 2020-05-05:
Also, I do not know how you have the things declared, of course, but please note the
`    .prog  = user_provided_block_device_prog,`
You have the & operator in front of the assignment
`    .read = &lfs_read,`

keck-in-space, 2020-05-05:
That's right. I'm getting a function pointer to my ```lfs_read``` function that's declared elsewhere.

thrasher8390, 2020-05-08:
@keck-in-space, when you call `lfs_ssize_t lfs_file_read(lfs_t *lfs, lfs_file_t *file,
        void *buffer, lfs_size_t size)` what is the size of your `buffer`? If that is sized at 4096 then lfs should be reading straight into that buffer. 

Can you debug your code and verify what `buffPtr` is being passed down to your block device (bd) when the 4096 read request occurs? It should be the pointer to the `buffer` that was passed into `lfs_file_read`. If you have a breakpoint in your bd you should see the `buffPtr` changing from your `read_cache` to the `buffer`. 

Again, I'm just speculating but that would be the test I would perform to ensure that there is a real concern with v2.2.1.

keck-in-space, 2020-05-08:
@thrasher8390 That's a good test. I'll keep it on my radar to test when possible. For now I've reverted to v2.1.4, but I too would like to know if this is a real issue in 2.2.1.

thrasher8390, 2020-05-08:
@keck-in-space, I just checked out the code and there was a change in `lfs_bd_read` it used to always read straight into the read_cache but now can read straight into the `buffer`

Old Code
[image]

New Code
https://github.com/ARMmbed/littlefs/blob/v2.2.1/lfs.c#L74-L81

ajaybhargav, 2021-05-24:
I think I have having problem which could be related to this. I am statically allocating buffers in my LFS configuration and they all sit next to each other.

 .bss._lfs_prog_buffer
 .bss._lfs_read_buffer
 .bss.lfs_enable_flag

somehow lfs_enable_flag which is 1 when LFS is successfully mounted (which I check it sets properly) but as read/write operation happens this flags becomes 0. in my code I am not clearing this flag anywhere except unmount which I am not calling at all.
Could it be that read buffer is being read over and above the allocated size?

In LFS configuration, I set:
cfg.read_size = page_sz;
cfg.prog_size = _cfg.read_size;
cfg.cache_size = LFS_CACHE_SZ; // LFS_CACHE_SZ = 1024
cfg.read_buffer = _lfs_read_buffer; // each buffer is LFS_CACHE_SZ
cfg.prog_buffer = _lfs_prog_buffer;