littlefs-project/littlefs #421: info about littlefs file limit on NAND

kazola, 2020-05-03:
Hi all,

I keep testing littlefs on Winbond W25N01GV NAND. 1024 sectors, 2048-byte pages, 64 pages / sector.

I low-level format the memory every time. Then I format with littlefs and do a test for 100 files (10 pages long so 20K in size) consisting in mount / file creation / writing / closing / dismount for each.

My .read_size, .prog_size and .cache_size are set to same value. Depending on it, littlefs can create more or less files and takes more or less time:

- 256 B, 9 files, ~6 seconds each file.
- 512 B, 20 files, ~4.5 seconds each file.
- 1024 B, 43 files, ~3.3 seconds each file.
- 2048 B. 100 files, ~variable, can take from 3 to 200 seconds, 40s, 15s, etc.

They seem proportional results up to a point. Even if it is an absurd case for NAND, I really cannot understand:

1. the file limit (the memory is huge, why only 9 files in case of 256 B?) and 
2. the variable time in the 2048 case.

Have you all a nice day.

thrasher8390, 2020-05-04:
Variable time is typically seen when lfs is compacting the meta-data pairs. Another compaction dependency that I've found is the # of files in a given directory. The more files in a single directory the longer it takes to compact that directory.

I'm not sure I follow your test setup though. You mentioned writing 100 20k files but then you call our 4 different size files. Can you explain your comment about the file limit and the problem that you are having? Are you saying that after righting 9 256byte files you get a failure during `lfs_file_write()`?

kazola, 2020-05-05:
Hi @thrasher8390 ,

thanks for your variance explanation! However, aren't 200 seconds simply too much, so something else may be happening? I have seen several posts / issues with similar problems.

About the tests. The test always is the same and consists of creating these 100 20k files. However, I first build and run the test with read_size, .prog_size and .cache_size = 256 bytes. Then I change this value to 512, I compile again and run. Then again with 1024, then again with 2048. Yes, for the 256 bytes value, I get a lfs_file_write error() after being able to create 9 files.

I really don't understand what may be happening here! Thanks for pointing out what wasn't clear, let's see if we can find out the problem / issue here :)

Have all a nice day.

visakh-png, 2020-05-07:
> Hi all,
> 
> I keep testing littlefs on Winbond W25N01GV NAND. 1024 sectors, 2048-byte pages, 64 pages / sector.
> 
> I low-level format the memory every time. Then I format with littlefs and do a test for 100 files (10 pages long so 20K in size) consisting in mount / file creation / writing / closing / dismount for each.
> 
> My .read_size, .prog_size and .cache_size are set to same value. Depending on such value, littlefs can create more or less files before launching an error on lfs_file_write(), when no more files are created, and takes more or less time:
> 
> * 256 B, 9 files, ~6 seconds each file.
> * 512 B, 20 files, ~4.5 seconds each file.
> * 1024 B, 43 files, ~3.3 seconds each file.
> * 2048 B. 100 files, ~variable, can take from 3 to 200 seconds, 40s, 15s, etc.
> 
> They seem proportional results up to a point. Even if it is an absurd case for NAND, I really cannot understand:
> 
> 1. the file limit (the memory is huge, why only 9 files in case of 256 B?) and
> 2. the variable time in the 2048 case.
> 
> Hope somebody can enlighten me! Have you all a nice day.

Hi I too have faced same issues performance and read/open failure, but not a write failure(When lookahead size is large) which is mentioned in  #422 . With this performance we cannot consider it anyhow. A question which out from the discussion is  about the NAND chip we uses. The data sheet says about a 128K erase  from the page address with D8-Block erase command. But I personally observed is , this command only erases 2KB page. Sio I have given lfs_config the block size equals to page size.  What is your opinion ?. Some posts says that for other variants they have successful use of lfs  #11 . But anything not mentioned about speed or other issues

kazola, 2020-05-07:
Oh, nice solution, although seems risky. Maybe we could contact Winbond and ask about this 2K block size of D8 command? Maybe by just setting the first page to all 0xFF they mark the whole block as available to save time? Cannot know.

Just being curious, does littlefs works for you this way by, for example, creating 40 files? From my experience, don't stop at 2, 3 files, it always work fast with the first ones.

visakh-png, 2020-05-07:
@kazola , yes lfs works for me, but with this heavy lag in performance. But when I tried to write 1MB file, reading back it throws error in fileopen itself. But a file up to 300K works perfect.

Regarding block erase I have asked questions to winbond.

kazola, 2020-05-08:
Perfect, hope you can ask what they answer here :) Thanks!

ryankurte, 2020-07-09:
heya, 

also observing reliability issues with writing files > a given size limit on an *25 flash IC, as well as with truncate (perhaps unsurprising for a flash fs) and with writing attributes to large files. 

we've done a bunch of testing of different configurations, without any real success. interestingly it seems correlated to a bunch of things. _i cannot for the life of me_ reproduce it using an in-memory driver (ie. without physical flash), so it's entirely possible that it is an issue with the driver, but we have reasonably exhaustively tested the behaviours of this too :-/

i suspect this is not useful for diagnostics, but, in case anyone else is in the same situation, our observations are:

- when writing files over a certain size threshold (~200KB on a 4MB file system), the file will either not be written, or the filesystem will become corrupt
  - decreasing / limiting file sizes appears to reduce the incidence of this, as for does decreasing the amount of available flash
  - when below the threshold, writing a second file of similar size (even when there is plenty of space available) also fails
- when the file system gets close to being full, often `LFS_ERR_CORRUPT` will occur rather than the expected zero sized write then `LFS_ERR_NOSPC` on sync
  - if the flash size is artificially limited (to ~256KB) the file system operates as expected
- when writing data in separate chunks, repeated `sync`ing of files appears to increase incidents of file system corruption
  - removing `syncs` increases the likelihood of success
- writing attributes to large files often fails (usually indicating the file does not exist), and corrupts the filesystem
- using `O_TRUNC` often fails, corrupting the filesystem
- running via asan while performing large file operations it is _sometimes_ possible to provoke overflows and other memory faults

the really tricky thing is that almost everything (except truncate, which again, is completely fair) works so well _up to a threshold_, and once this is passed it all seems to fall apart. if it is a driver issue it is a bit hard to imagine why it would be impacted so significantly by tuning file system configuration and write-patterns.

visakh-png, 2020-11-17:
> Perfect, hope you can share what they answer here :) Thanks!

Hi @kazola , I have stopped that project in the middle and re-opened.  There was a problem with the QSPI driver. This was not sending addresses as expected through commands or this was not expected by the NAND chip. The address is now sending as data form to solve this issue and it is rectified. But now I not uses LFS, instead I  keep data in specific addresses  and using a static queue to identify which and where , and of course for wear-levelling. 

Note: I will try to use LFS again in near future and will share observations here. 

Regards,
visakh

closed 2021-11-30 by kazola