littlefs-project/littlefs #422: File open fails for large files and performance glitch in read for littlefs with eMMC NAND W25N01GV

visakh-png, 2020-05-05:
Hi,
I am doing feasibility of littlefs in my STML4 controller with an external QSPI flash W25N01GV.

1. Writing of 100K in 10K chunks is not taking considerable time(~10S), but reading back it takes around ~30 Seconds. When it is around 300K the read time is 1.45 Minute. I wish to confirm is this behavior is expected.(Note: my raw functions for  quadSPI flash  read and write completes within a second)

2. The other issue I have observed is after writing large data to  files(>300K) the fileopen operation for read got crashes most of the time.

 Below is my configuration , please check and add your comments/Solution/Suggestion.

	cfg.read_size 		= W25N01GV_PAGE_SIZE; //2K
	cfg.prog_size 		= W25N01GV_PAGE_SIZE;
	cfg.block_size 		= W25N01GV_PAGE_SIZE;//*W25N01_PAGE_BLOCK_COUNT;
	cfg.block_count 	= W25N01GV_PAGE_COUNT; //65535
	cfg.lookahead_size 	= W25N01GV_PAGE_SIZE;
	cfg.cache_size 		= W25N01GV_PAGE_SIZE;
	cfg.block_cycles 	= BLOCK_CYCLES_TO_READ; //500

C47D, 2020-05-05:
Seems very similar to #421

Dothrakki, 2020-09-03:
Hi  visakh-png ,

I am a bit curious about how you have implemented the erase function?
since the flash chip block size is 64 pages and total number of available blocks are 1024, how are you handling the erase block command?
in my understanding the minimum erasable unit is 1 block, which is 64 pages for w25n01gv. (correct me if i am wrong)
in your configuration, you have 65535 blocks and block size is 2k, how are you erasing 2k of data? wouldn't it erase the complete hardware block? (ie 64 pages)
Have you crosschecked the data integrity of your file which is 300k ? i am running into some data corruption while increasing the file size.

visakh-png, 2020-11-17:
hi @Dothrakki .

Yes. I have some issue with my QSPI driver. When resolved it, as u said it will erase 64 pages