littlefs-project/littlefs #425: Calling lfs_file_write() until you prompt 'No more free space' causes a strange problem

LusterWong, 2020-05-12:
**Two things have been found so far：**

1. Keep executing lfs_file_write() until it prompts 'No more free space', then execute lfs_file_truncate(LFS, file, 0) to empty the current file, and then call lfs_file_write() again, which will also prompt the error of 'No more free space'.
It is effective to empty the file, because lfs_file_size() will be called first to get the file size, and lfs_file_truncate(LFS, file, 0) will be called again after lfs_file_size() to confirm that the file size is 0.Do not understand why there will be a lack of space prompt.
Because there is No more free space, the solution is to keep lfs_file_write() from filling the file system, but to do so anticipate the boundaries of the space.But that would be foolish and inefficient.

2. Keep executing lfs_file_write() until it prompts 'No more free space', then shut down the system, re-power into the system to read the data just written, found no data can be read, file size is 0.
Similar to the first problem, this happens whenever you are prompted 'No more free space' when writing.The current solution is to call lfs_file_sync() after each successful call to lfs_file_write(), so that the data can be read correctly if there is not enough space to write to the prompt.However, this is also silly, the write speed becomes very slow, and the caching mechanism is worthless.

--------------------------------
Maybe the above problem is caused by the mechanism of littlefs that I don't know much about, but I have been debugging for a long time and can't find the reason. Please help me, thank you

Johnxjj, 2020-05-26:
Have you checked or set the cursor position at the beginning before rewriting the file after truncating the file?

LusterWong, 2020-05-27:
> truncating

Yes. Because I just want to clear the file, change the file size to 0 instead of deleting the file, so after clearing (truncat) is finished, set the file pointer back to the beginning of the file before writing.
In short, at present, once a file is full in any write operation, an exception occurs and it has not been resolved. The effect is the same whether it is external FLASH or MCU internal FLASH.

Johnxjj, 2020-05-27:
I think you should call lfs_file_sync() after "No more free space", and then continue with other operation, it should be normal.

LusterWong, 2020-05-29:
> I think you should call lfs_file_sync() after "No more free space", and then continue with other operation, it should be normal.

`bool FileWrite(File FileHandle, const void* Data, uint32_t Size)
{
	if(lfs_file_write(&FS, &FileHandle->LFFile, Data, Size) > 0)
	{
		lfs_file_sync(&FS, &FileHandle->LFFile);
		return(TRUE);
	}
	lfs_file_sync(&FS, &FileHandle->LFFile);
	return(FALSE);
}

bool FileClear(File FileHandle)
{
	if(lfs_file_truncate(&FS, &FileHandle->LFFile, 0) < 0)
	{
		lfs_file_sync(&FS, &FileHandle->LFFile);
		return(FALSE);
	}
	lfs_file_sync(&FS, &FileHandle->LFFile);
	return(TRUE);
}`
The code looks like this, and I'm repackaging it.
As you suggested, plus the call to lfs_file_sync after writing full, the data was written successfully and read out, but when I tried to empty the file (call FileClear, and reset the file pointer back to 0) and write again, I continued to get a write error.

Johnxjj, 2020-05-29:
You can try it as this. open->write->sync->close.    open->truncate->close.   open->write-sync->close

LusterWong, 2020-05-29:
> You can try it as this. open->write->sync->close. open->truncate->close. open->write-sync->close

It's normal to turn it off and back on, and that's the solution that's currently in use, as I said at the beginning.
However, I am worried about the heap overflow caused by opening and closing, and the memory space used by MCU is limited