littlefs-project/littlefs #426: A little problem with the look-ahead cache？

Johnxjj, 2020-05-12:
`
static int lfs_alloc_lookahead(void *p, lfs_block_t block) {
    lfs_t *lfs = (lfs_t*)p;
    lfs_block_t off = ((block - lfs->free.off)
            + lfs->cfg->block_count) % lfs->cfg->block_count;

    if (off < lfs->free.size) {
        lfs->free.buffer[off / 32] |= 1U << (off % 32);
    }

    return 0;
}
`
     I encountered a problem while reading the code, can anyone help me?
Why is it necessary to add lfs-> free.off (a random number calculated by CRC) to the lookahead_buffer instead of recording the usage of each block in sequence.

     In addition，
**lfs-> free.i** Does it refer to the number of free blocks calculated after this power-on？
**lfs-> free.ack** I found that it sometimes resets, which makes me unclear what it means. Can anyone explain what it means？

geky, 2020-07-17:
Quick notes:
 
> Why is it necessary to add lfs-> free.off (a random number calculated by CRC)

littlefs does this for wear leveling. It does traveling across the disk linearly after each boot, but it doesn't store where it was when power is lost (I think this is something that may need to change in the future).

So instead, every boot it chooses a random offset to start block allocation from.

This should approach a uniform distribution of blocks sufficient enough for providing wear-leveling across boots.

Otherwise, you risk wearing out the first couple of blocks if you start with those every time you reboot the device.

geky, 2020-07-17:
Oh, then I realized I didn't answer your other questions...

> lfs-> free.i Does it refer to the number of free blocks calculated after this power-on？

The block allocator keeps a bit-vector containing information on which blocks are free/in-use (this is the "lookahead buffer"). lfs->free.i is the offset into that bit-vector. We allocate from that bit-vector in order.

> lfs-> free.ack I found that it sometimes resets, which makes me unclear what it means. Can anyone explain what it means？

lfs->free.ack is a bit complicated... The normal block allocator is designed to handle wrap-arounds. That is, if the lfs->free.i reaches the end of the address space, it will keep scanning from block 0 to look for new blocks.

But then how do we know if we run out of blocks? We used lfs->free.ack ("acknowledge block" shortened) to keep track of blocks that have not only been allocated, but have been written to successfully and committed as part of the filesystem. If an operation is successful, it calls lfs_alloc_ack to indicate all the allocated blocks are a part of a successful commit. 

Sometimes littlefs needs to allocate multiple blocks. "metadata-pairs" need two blocks for example. So if we're writing a metadata-block, we won't "lfs_alloc_ack" until we have two successful allocations and have managed to write to them without errors.

If we ever scan the entire filesystem and lfs->free.i touches lfs->free.ack, that means that the current operation wasn't able to complete in the blocks we could find, so we return LFS_ERR_NOSPC.

I realize this part is really complex and I'm trying to answer with little time. Feel free to ask questions and I'll try to explain it better next chance I get.

Johnxjj, 2020-07-17:
Thank you very much for your answer !

closed 2023-11-09 by Johnxjj