littlefs-project/littlefs #427: thread-safety

PoppaChubby, 2020-05-13:
Hello, and congratulations for this great project

I successfully integrated littlefs into my project quite quicly, many thanks for the great documentation and all the usefull information available here.
For information I use a NOR flash, an LPC µC, and freeRTOS. I configured lfs like this :
.read_size = 1
.prog_size = 1
.block_size = 4096 (my flash sector size)
.block_count = FLASH_CHIP_SIZE / 4096
.cache_size = 64
.lookahead_size = 64
.block_cycles = 500,

As my project is multi-tasked and several tasks can make use of files, I initially protected the flash access by a mutex, in each of the "user_provided_block_device_" function.

And after that I found - and applied - advices from these messages:
#https://github.com/ARMmbed/littlefs/issues/156
#https://github.com/ARMmbed/littlefs/pull/317

I managed simultaneous open files with static alloc using lfs_file_opencfg, as explained in :
#https://github.com/ARMmbed/littlefs/issues/117
And I also added a bound on max simultaneous opened files.

So I have a recursive mutex that protect the calls of each lfs function. The question I would have is : do I need to keep my mutex in the "user_provided_block_device_" functions, or is it totally useless, now that I have another one protecting each lfs call? As all my "user_provided_block_device_" functions are blocking, so I would say I can remove the first mutex, but I am not really sure...

e107steved, 2020-05-14:
I run with a mutex protecting each lfs call, and nothing on the block device, which seems to work fine; logically its protecting at a higher level. (But I do have a low-level mutex where the block device is shared between lfs and other purposes, simply to manage comms)

PoppaChubby, 2020-05-14:
OK, thanks. That's what I thought but I am always nervous with multi-tasking...

closed 2020-05-15 by PoppaChubby