littlefs-project/littlefs #428: lfs_mkdir() returns LFS_ERR_NOENT

PoppaChubby, 2020-05-15:
Hello,

my attempt to create a dir fails with LFS_ERR_NOENT... Of course there is no directory entry, or I would not been trying to create one !

```
/// Top level directory operations ///
int lfs_mkdir(lfs_t *lfs, const char *path) {
    LFS_TRACE("lfs_mkdir(%p, \"%s\")", (void*)lfs, path);
    // deorphan if we haven't yet, needed at most once after poweron
    int err = lfs_fs_forceconsistency(lfs);
    if (err) {
        LFS_TRACE("lfs_mkdir -> %d", err);
        return err;
    }

    struct lfs_mlist cwd;
    cwd.next = lfs->mlist;
    uint16_t id;
    err = lfs_dir_find(lfs, &cwd.m, &path, &id);
    if (!(err == LFS_ERR_NOENT && id != 0x3ff)) {
        LFS_TRACE("lfs_mkdir -> %d", (err < 0) ? err : LFS_ERR_EXIST);
        return (err < 0) ? err : LFS_ERR_EXIST;
    }
(...)
```
This test is failing in lfs_mkdir() because id ==0x3ff. I have no idea of the meaning of id, and why lfs_dir_find() puts  0x3ff in id. 
It seems that I missed something obvious... Do I need to call something before lfs_mkdir() ? (of course I mounted successfully the filsystem...)

Thanks for your help

PoppaChubby, 2020-05-18:
OK, problem solved, sorry for the noise.
My error was to try to create "myDir/". With "myDir" without the trailing "/", lfs_mkdir() returns 0.

- is there a doc somewhere explaining requirements on file and directory names ?
- what is the reason for returning LFS_ERR_NOENT when I try to create "myDir/" ? Does it try to create "" under "myDir" and complains because "myDir" does not exists ?

pavel-kirienko, 2021-07-01:
I observe the same behavior: `lfs_mkdir(..., "/a/")` fails with `LFS_ERR_NOENT`, but `lfs_mkdir(..., "/a")` succeeds. I'm not sure it's a defect, though. If you look at the standard, it says that ENOENT shall be returned if:

> A component of the path prefix specified by path does not name an existing directory or path is an empty string.

If you decompose `/a/`, the component `a` names a non-existent directory, so the function is expected to return ENOENT. Either that or I am misreading the standard.

inf265, 2022-10-11:
Same for me. Searched a while why my mkdir is not working. At least it was the trailing "/" and the end of the string.

geky, 2022-10-24:
https://github.com/littlefs-project/littlefs/pull/679 is related.

This is a tricky part of path parsing where following the logic (mkdir "a/" => mkdir "" in "a") does not match what POSIX and most filesystems do.

I'll make sure this is fixed next minor release.

geky, 2024-11-24:
> I'll make sure this is fixed next minor release.

I may have been slightly wrong in this time estimate,

But this should now be implemented in https://github.com/littlefs-project/littlefs/pull/1046. Feedback welcome.