littlefs-project/littlefs #429: infinite loop during file close

Luka234, 2020-05-17:
if lfs_file_open is called two times (with identical arguments) it will return ok to both calls. That will create self referencing linked list (lfs->mlist->next will look to itself). During file closing (in function lfs_dir_commit) linked list will create infinite loop and never return.

I'm aware that described usage is not valid but maybe you could add assertion or return error result on second call. I discovered that behavior when I've tested lfs_file_opencfg bug forgot to remove previous lfs_file_open.

NewbieYangbo, 2020-05-25:
I met similar problem when i repeat to  create-write-close new file  use timer.
when file numbes are more than 200+, this problem will occurs.

MarkusK92, 2020-06-15:
@Luka234 Thanks for sharing your knowledge! It saved me many days of debugging and so much pain!

I solved the problem by putting an additional layer on top of lfs_file_open_cfg and checking if the LFS_F_OPENED flag is already set by the last file_open request:

Example code snippet just to get what I mean:
`if (file.flags & LFS_F_OPENED) {`
`    // File is already open. Skip this file_open request`
`} else {`
`    lfs_file_opencfg(&lfs, &file, path, LFS_O_RDWR | LFS_O_CREAT, &file_cfg);`
`}`

The same thing I did with file_close! A second file close leads to problems as well.

NoahGorny, 2020-06-17:
I'm trying to create a PR with an assert on lfs_file_opencfg, but lfs_file struct may be uninitialised...
I'm thinking about creating a magic member/ validating someway that the struct is initialized and the file is already opened

PR #443 (2020-12-04): Assert that the file isnt open in lfs_file_opencfg
Closes #429

Also fixed a magic number in `opencfg`