littlefs-project/littlefs #433: read access slowing down boot after boot

PoppaChubby, 2020-05-27:
Hi,
I noticed a strange behaviour that may be normal but I want to be sure.

I run the following code:
- example code (boot count) : so read + write access
- benchmark uC cycles start
- file read of another file not invloved in boot count
- benchmark uC cycles stop

My initial intention was to optimize the file read (I implemented a fgets() function, and I wanted to compare a byte by byte implementation to a buffered implementation).
I noticed that my processor cycles count is always incrementing boot after boot (~ 6000 cycles more each time). If I remove the example code, my processor cycles count is stable after each boot. 

I understand that the writes change the file system, so that can change the speed of some operation and behaviour of garbage collection. 
But is it normal that the speed of reading a file that has not been touched changes ? And will it stop changing one day ?

Roger-Waters, 2020-06-18:
Confirm. I have met the same issue.

thrasher8390, 2020-08-19:
@blomnik and @PoppaChubby,

What flags are you using when you open the boot count file? If you open the file as read/write the file system will always commit a new version of the file which could result in compaction. 

I recommend opening the file as read only if you don't intend on altering the file content.

geky, 2020-08-20:
Hi, sorry about the late response. This sounds like the files are sharing a metadata pair.

In littlefs, each file's data is stored in a skip-list of blocks. These lists are independent, so a write to one file doesn't affect the other file. But this isn't true for the _metadata_ of a file (the file type, it's length, the pointer to the skip-list).

In littlefs a file's metadata is stored in metadata pairs. These are 2-block logs that provide atomic updates by crcing new metadata which is appended instead of overwritten. But giving each file its own metadata pair turns out to be rather expensive, especially for small files, so instead multiple files in the same directory have their metadata placed in the same metadata pair.

One of the current issues right now (performance wise) with littlefs is that these metadata logs need to be scanned to find the most recent piece of metadata. This starts off cheap when the metadata logs are first initialized, but as the metadata logs fill with updates this scanning cost grows until the metadata pair is full and metadata compaction (in-log garbage collection) kicks in.

And, as a side-effect of files sharing metadata pairs, writes to one file can cause a shared metadata pair to fill with updates.

---

Another interesting thing is that metadata pairs in littlefs are closely tied to directories. Directories are just linked-lists of metadata pairs which get scanned for filenames when you open a file (the filename is just another piece of metadata).

This means you could store the two files in different directories to force independent metadata pairs.
``` c
lfs_file_open(&lfs, &file, "/readonly_files/readonly_file_here.txt", LFS_O_RDONLY);
lfs_file_open(&lfs, &file, "/writeable_files/writable_file_here.txt", LFS_O_RDWR);
```

Not necessarily the nicest solution, but it is an option.

blomnik, 2020-08-20:
Hello, everybody.
@thrasher8390 , if you will look at the 'boot_count' example in README.md file, you'll get an answer for your post: it is impossible to open this file for read only. :)

@geky, I guess it is a source of performance degradation. Thank you for your response. Unfortantly, I can't split READONLY and READWRITE modes for 'boot_count' file due to obvious reasons. :) But in all other cases, it useful to know this "trick". I guess, this design was described in DESIGN.md, but it is not obvious that metadata (is it tags?) grows until GC compact it.

geky, 2020-08-20:
Yeah, the "tags" are pieces of metadata written to the log. The theory is that the growth of the metadata is bounded to one erase block, so it doesn't grow indefinitely unlike in true logging filesystems. But with large erase blocks it becomes significant.

At some point I should go through and add some illustrative examples of how littlefs's internals work. But that may have to come after taking care of the other issues.

katmorris88, 2021-06-22:
We are experiencing this same issue. With every file commit open/read/write/close operations gradually increase in time until the metadata block becomes full and garbage collection is finally run. It's undesirable in our system to have to wait for an undetermined amount of file writes until lfs_dir_compact is called in which we can see up to a 4s delay while in the process of writing a file. Would it be okay to run lfs_dir_compact on a directory in the background as a low priority task so that we can keep a more deterministic time when writing files to the filesystem?

geky, 2021-06-23:
Just an FYI, but since this issue was posted @mon has added a workaround in the form of a `metadata_max` config option (https://github.com/littlefs-project/littlefs/pull/502):
[https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.h#L264 ](https://github.com/littlefs-project/littlefs/blob/ead50807f1ca3fdf2da00b77a0ce02651ded2d13/lfs.h#L264)

This works around the metadata scalability issues by, well, keeping the metadata from scaling. I suspect setting `metadata_max < ~1024` should help limit this issue at the cost of wasting parts of the block.

> Would it be okay to run lfs_dir_compact on a directory in the background as a low priority task so that we can keep a more deterministic time when writing files to the filesystem?

This isn't a bad idea, such a function could also take a threshold argument, so that `lfs_dir_compact` is only run on directories that are, say, 75% full.

I believe this is captured in #523, but feel free to add any comments/details/ideas:
https://github.com/littlefs-project/littlefs/issues/523

Feel free to create a PR if anyone wants to take a stab at it. In my current TODO list I'm hoping to take a wider look at the allocator and its performance, but it won't be for a while, at least not until after looking at how to speed up the metadata itself.