littlefs-project/littlefs #434: File size getting added after each 64 characters

ByteGalaxy, 2020-06-01:
Hello,

I am writing a set of characters in one file and reading them back.
After reading, I have found that the read size is getting added  in the file after each 64 characters.
 eg. I am writing 0x31, 0x32,.....,0x38 multiple times unless file size is 2KB
After reading, I am getting this,

 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0xC0 0x07 0x0A
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x31 0x32 0x33 0x34 0x35 0x36 0x37 0x38
 0x0A

Has anyone came across this situation?
How you have fixed it?

e107steved, 2020-06-03:
It seems extremely unlikely that this is a littleFs problem - if you look at the number of people successfully using littleFs, and the detailed testing others have done, a fundamental problem like this would surely have been found already.
So I suggest you have a very close look at your code, and if you can't see the problem then add details of your configuration and code here. 
The "0x0A" value suggests that maybe you are writing data using writeln() or similar, since its an end of line character.

ByteGalaxy, 2020-06-04:
@e107steved : My bad..
I was adding a read length into the print buffer using some other function.
Now I have fixed it.

closed 2020-06-04 by ByteGalaxy