littlefs-project/littlefs #435: Use of LFS_O_EXCL Flag

FACTSMC, 2020-06-03:
I have been testing out littlefs for a while now and have updated to latest version (V2.2). I want to use LFS_O_WRONLY | LFS_O_EXCL as my flags to have exclusive write permissions for a created file. I tried this and got back LFS_ERR_EXIST(-17). Upon looking further into lfs.c, LFS_O_EXCL is only used in one place and that is to check if the flag is set and mark it as an error. 

I am under the impression that LFS_O_EXCL was used for file exclusivity. I just want to make sure I am using this flag correctly.

geky, 2020-07-17:
Hi @FACTSMC, littlefs follows fairly strictly the behavior of the POSIX definitions (they came first and it's what a large group of developers expect even if it is _extremely_ quirky):
https://pubs.opengroup.org/onlinepubs/009695399/functions/open.html

LFS_O_EXCL just indicates that lfs_file_open should error if the file exists on disk. On Linux/Unix systems files are often used for synchronizations ("dpkg.lock" files for example), so that's where the name comes from.

Unfortunately littlefs doesn't really track the files that are open in a way that would let you limit an application to opening a file once. The best solution I can think of would be to manage this in your application.

FACTSMC, 2020-07-20:
Hi @geky, Thank you for the response. This was my conclusion, but wanted to verify.

closed 2020-07-20 by FACTSMC