littlefs-project/littlefs #436: Seeking to the current file offset is inefficient

embeddedt, 2020-06-04:
If you call `lfs_file_seek` with `LFS_SEEK_SET` and the current offset of the file, it still [unconditionally flushes](https://github.com/ARMmbed/littlefs/blob/4c9146ea539f72749d6cc3ea076372a81b12cb11/lfs.c#L3009) the file.

In my case, I'm using littlefs with a custom VFS wrapper, so I reset littlefs' file offset each time I write to the file. This "feature" resulted in significant performance loss.

Currently, I've worked around this by calling `lfs_file_tell` before I set the new offset. I'd be happy to send a pull request that:

* moves the flushing to after the new file position is calculated, and
* adds a check for whether the new position will be the same

BrianPugh, 2020-11-11:
@embeddedt hey! i just noticed this issue, are you using my port by chance? If so, we can maybe make some tweaks there. Open an issue if you have any questions/concerns/whatever.

Cool seeing you here, sorry I haven't contributed to lvgl in a little bit :P

embeddedt, 2020-11-11:
Good to see you too. :slightly_smiling_face: 

If by port you mean https://github.com/joltwallet/esp_littlefs, then no. I was using a custom VFS system which controls the offsets itself. I'm not sure if ESP-IDF's system suffers from this problem or not, as I don't use ESP32.

geky, 2020-11-14:
Hi @embeddedt, thanks for creating a issue! sorry for the late response. I'd be happy to take in this change, assuming it still matches the POSIX semantics ([lseek](https://linux.die.net/man/2/lseek)).

One concern is that lseek is expected to be the sync-point where writes become reflected in read, in the case that you're switching between read/write operations in the same file. I'm trying to think of corner cases, but so far I can't think of any where this optimization wouldn't work. At the very least it would be good to have a test for read-after-write-with-seek-to-same-location if we don't already have one.

embeddedt, 2021-01-04:
@geky Not sure if you've seen #511 or not.

geky, 2021-01-06:
Hello! Yes, sorry, I haven't been able to look into it yet. It looks good! Will comment on #511.

PR #511 (2021-01-18): Skip flushing file if lfs_file_rawseek() doesn't change position
This fixes #436.

@geky It appears that your concerns were valid, as one of the truncation test cases is now failing with this change:

```
====== results ======
tests/test_truncate.toml:96:failure: test_truncate#3#1 failed with -6
tests/test_truncate.toml:127:assert: assert failed with 48, expected eq 0
    memcmp(rb, wb, trunc) => 0;
```

Do you have any ideas about how to go about resolving this?
closed 2023-07-10 by embeddedt