littlefs-project/littlefs #437: IAR ARM compilation errors

sevstels, 2020-06-05:
Error[Pe144]: a value of type "void const *" cannot be used to initialize an entity of type "uint8_t const lfs_util.c 22 

Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *" \bd\lfs_testbd.c 30 
Error[Pe513]: a value of type "void *" cannot be assigned to an entity of type "uint32_t *"  \bd\lfs_testbd.c 39 
Error[Pe513]: a value of type "void *" cannot be assigned to an entity of type "uint32_t *"  \bd\lfs_testbd.c 41 
Error[Pe029]: expected an expression  \bd\lfs_testbd.c 53 
Error[Pe029]: expected an expression  \bd\lfs_testbd.c 60 
Warning[Pe012]: parsing restarts here after previous syntax error  \bd\lfs_testbd.c 63 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 89 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 108 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 118 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 128 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 137 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 151 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_testbd.c 154 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 176 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_testbd.c 179 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 222 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_testbd.c 225 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 279 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_testbd.c 282 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_testbd *"  \bd\lfs_testbd.c 292 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_testbd.c 295 
Error while running C/C++ Compiler 
lfs_filebd.c  
Warning[Pa050]: non-native end of line sequence detected (this diagnostic is only issued once)  \bd\lfs_filebd.h 1 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_filebd *"  \bd\lfs_filebd.c 26 
Error[Pe020]: identifier "O_RDWR" is undefined  \bd\lfs_filebd.c 30 
Error[Pe020]: identifier "O_CREAT" is undefined  \bd\lfs_filebd.c 30 
Error[Pe020]: identifier "open" is undefined  \bd\lfs_filebd.c 30 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_filebd *"  \bd\lfs_filebd.c 60 
Error[Pe020]: identifier "close" is undefined  \bd\lfs_filebd.c 61 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_filebd *"  \bd\lfs_filebd.c 76 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_filebd.c 79 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 89 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 90 
Error[Pe018]: expected a ")"  \bd\lfs_filebd.c 90 
Error[Pe020]: identifier "lseek" is undefined  \bd\lfs_filebd.c 89 
Error[Pe040]: expected an identifier  \bd\lfs_filebd.c 90 
Error[Pe020]: identifier "ssize_t" is undefined  \bd\lfs_filebd.c 97 
Error[Pe020]: identifier "read" is undefined  \bd\lfs_filebd.c 97 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_filebd *"  \bd\lfs_filebd.c 112 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_filebd.c 115 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 121 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 122 
Error[Pe018]: expected a ")"  \bd\lfs_filebd.c 122 
Error[Pe020]: identifier "lseek" is undefined  \bd\lfs_filebd.c 121 
Error[Pe040]: expected an identifier  \bd\lfs_filebd.c 122 
Error[Pe020]: identifier "ssize_t" is undefined  \bd\lfs_filebd.c 131 
Error[Pe020]: identifier "read" is undefined  \bd\lfs_filebd.c 131 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 143 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 144 
Error[Pe018]: expected a ")"  \bd\lfs_filebd.c 144 
Error[Pe020]: identifier "lseek" is undefined  \bd\lfs_filebd.c 143 
Error[Pe040]: expected an identifier  \bd\lfs_filebd.c 144 
Error[Pe020]: identifier "ssize_t" is undefined  \bd\lfs_filebd.c 151 
Error[Pe020]: identifier "write" is undefined  \bd\lfs_filebd.c 151 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_filebd *"  \bd\lfs_filebd.c 164 
Error[Pe020]: identifier "assert" is undefined  \bd\lfs_filebd.c 167 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 171 
Error[Pe020]: identifier "off_t" is undefined  \bd\lfs_filebd.c 171 
Error[Pe018]: expected a ")"  \bd\lfs_filebd.c 171 
Error[Pe020]: identifier "lseek" is undefined  \bd\lfs_filebd.c 171 
Error[Pe040]: expected an identifier  \bd\lfs_filebd.c 171 
Error[Pe020]: identifier "ssize_t" is undefined  \bd\lfs_filebd.c 179 
Error[Pe029]: expected an expression  \bd\lfs_filebd.c 179 
Error[Pe020]: identifier "write" is undefined  \bd\lfs_filebd.c 179 
Error[Pe144]: a value of type "void *" cannot be used to initialize an entity of type "lfs_filebd *"  \bd\lfs_filebd.c 195 
Error[Pe020]: identifier "fsync" is undefined  \bd\lfs_filebd.c 196 
Error while running C/C++ Compiler 
[... 86 lines cut ...]
Error[Pe2363]: cast to incomplete array type "lfs_mattr []" is not allowed  \lfs.c 1977 
Error[Pe029]: expected an expression  \lfs.c 1977 
Error[Pe2363]: cast to incomplete array type "lfs_mattr []" is not allowed  \lfs.c 1977 
Error[Pe029]: expected an expression  \lfs.c 1977 
Error[Pe2363]: cast to incomplete array type "lfs_mattr []" is not allowed  \lfs.c 1992 
Error[Pe029]: expected an expression  \lfs.c 1992 
Error[Pe2363]: cast to incomplete array type "lfs_mattr []" is not allowed  \lfs.c 1992 
Error[Pe029]: expected an expression  \lfs.c 1992 
Error[Pe513]: a value of type "lfs_mlist *" cannot be assigned to an entity of type "lfs::lfs_mlist *"  \lfs.c 2053 
Error[Pe144]: a value of type "lfs::lfs_mlist **" cannot be used to initialize an entity of type "lfs_mlist **"  \lfs.c 2062 
Error[Pe393]: pointer to incomplete class type is not allowed  \lfs.c 2062 
Error[Pe393]: pointer to incomplete class type is not allowed  \lfs.c 2064 
Error[Pe029]: expected an expression  \lfs.c 2216 
Warning[Pe224]: the format string requires additional arguments  \lfs.c 2331 
Error[Pe018]: expected a ")"  \lfs.c 2331 
Error[Pe029]: expected an expression  \lfs.c 2346 
Error[Pe513]: a value of type "lfs_mlist *" cannot be assigned to an entity of type "lfs::lfs_mlist *"  \lfs.c 2418 
Error[Pe2363]: cast to incomplete array type "lfs_mattr []" is not allowed  \lfs.c 2434 
Error[Pe029]: expected an expression  \lfs.c 2434 
Error[Pe2363]: cast to incomplete array type "lfs_mattr []" is not allowed  \lfs.c 2434 
Error[Pe029]: expected an expression  \lfs.c 2434 
Error[Pe167]: argument of type "lfs_file::lfs_ctz *" is incompatible with parameter of type "lfs_ctz *"  \lfs.c 2462 
Error[Pe513]: a value of type "void *" cannot be assigned to an entity of type "uint8_t *"  \lfs.c 2491 
Error limit reached 
Error while running C/C++ Compiler

geky, 2020-07-17:
I'm not really sure how to respond to this. Looking through a few of the warnings and the existing code is perfectly valid c99 code.

Are you sure you have IAR configured to compile in c99 mode?
https://www.iar.com/support/resources/articles/exploring-c11-and-c14/

sevstels, 2020-07-18:
If change the compilation mode, then the rest of the project wrong building. The principle "Either this or that".

I almost rework littlefs for C ++ without using macros. Macros are real Evil.
But unfortunately I could not correctly replace macros:

LFS_MKATTRS
LFS_MKTAG

 with inline functions.

If you can help me with this, then littlefs will build correctly in all cases. Thnk.

geky, 2020-07-18:
Ah, unfortunately C++ is no longer a superset of C. As they bother developed they have diverged. I don't know the history there, but the version of C compatible with C++ (C89 I think?) lacks a lot of usability features (variables must be initialized at start of block, static, const, no C++ comments (//)), so I believe C99 is the best C standard to adopt at this time.

To combine C++/littlefs I would use two different compilation modes:
gcc -std=c99 lfs.c -o lfs.o
g++ -std=c++11 main.cpp -o main.o
g++ lfs.o main.o -o app.elf

> If change the compilation mode, then the rest of the project wrong building. The principle "Either this or that".

That sounds like an IDE/build system limitation. What IDE/build system are you using? Can it not handle .c and .cpp files differently? 

> I almost rework littlefs for C ++ without using macros. Macros are real Evil.
> But unfortunately I could not correctly replace macros:
>
> LFS_MKATTRS
> LFS_MKTAG

LFS_MKATTRS creates an array on the stack which is used to pass a LISP-like list of commands to lfs_dir_commit. This is one case where an inline function won't work as it needs a variable number of arguments.

It does this using compound literals, another feature in C99 but not in C++. I think the closest you can get is to create a class using parameter pack, but I don't know exactly what it would look like:
https://en.cppreference.com/w/cpp/language/parameter_pack

sevstels, 2020-07-21:
I use IAR for ARM v8.32

For my taste, using macros on the stack makes the code unpredictable. This is why the project has so many error messages. The C ++ standard is specifically designed to improve the reliability of software. And practice shows us that the littlefs code is not reliable and not universal. 

Later, I will probably complete a predictable stable version for C ++. When finished, I'll write a message here.

geky, 2020-11-14:
A C++ port would be very interesting, at least to compare the two.