littlefs-project/littlefs #439: the function lfs_file_flush does so much reiterated work?

chrome777, 2020-06-10:
```c

static int lfs_file_flush(lfs_t *lfs, lfs_file_t *file) {
    LFS_ASSERT(file->flags & LFS_F_OPENED);

    ......
            while (file->pos < file->ctz.size) {
                // copy over a byte at a time, leave it up to caching
                // to make this efficient
                uint8_t data;
                lfs_ssize_t res = lfs_file_read(lfs, &orig, &data, 1);
                if (res < 0) {
                    return res;
                }

                res = lfs_file_write(lfs, file, &data, 1);
                if (res < 0) {
                    return res;
                }

                // keep our reference to the rcache in sync
                if (lfs->rcache.block != LFS_BLOCK_NULL) {
                    lfs_cache_drop(lfs, &orig.cache);
                    lfs_cache_drop(lfs, &lfs->rcache);
                }
            }

    ......

    return 0;
}

```

### when we use lfs_file_seek to change the file->pos, it repeats reading and writing till the file->pos is equal to the file->ctz.size; any problems with my understanding?

### eg: we have already  created a 1MiB file, when we change the file->pos to the head, and close the file after writen a byte data 0x05, then the function lfs_file_seek takes effect, repeats 1M times to read and write, it may take a long time waiting, does this design reasonable? 

### Ignore my poor English. Hope to get some suggestions.  thanks

Johnxjj, 2020-06-12:
Modifying the header data will overwrite the following 1M data。

geky, 2020-07-17:
> then the function lfs_file_seek takes effect, repeats 1M times to read and write

This sounds like an issue with littlefs's append-only behavior. Unfortunately, random writes turn into rewriting the entire files.
https://github.com/ARMmbed/littlefs/issues/27#issuecomment-364992053

It's something I want to fix, but will take some time.