littlefs-project/littlefs #440: Pre-emptively mark a file to be not inline?

ancaritha, 2020-06-11:
If I know in advance that a file will be very large, is there a way to pre-emptively mark it to not be an inline file when I create it?

In our file system, we have a few small files used to store things like calibration data, but we know in advance things like the datalog will be multiple megabytes in size.  It'd be nice to be able to tell the system in advance that a file will be large so it reduces the amounts of reading/writing/moving around that is done before we hit that magic limit where it decides it's no longer inline.

geky, 2020-07-17:
This is a good idea, and could be added with a flag (LFS_O_NOINLINE?).

One question, inline files are < cache size, and files aren't written to disk until > cache size. Is this occurring over a number of small writes? Or do you think there might be a bug causing excessive disk operations?

ancaritha, 2020-07-20:
I don't think there's a bug or anything but neither have I looked in depth at the SPI traffic.  I have a little bit for some stuff but I was mostly writing large pieces of data at once (half a sector size) so after 2 writes I passed the INLINE threshold.  I haven't looked at it with this specific issue in mind (appending 20 to 40 bytes to a file at a time over an extended period of time).  I have a vague understanding of how LFS works but the details of it are too complex for me to fully grasp so I wasn't sure how often it was changing sectors when I was writing to a file that was inlined in a directory structure.  I'm not sure how often it has to update the 'directory' info when a file in said directory changes so I just thought it'd be nice to be able to say in advance "This file will be big".

geky, 2020-07-25:
Ah I see.

I don't think the performance improvements from LFS_O_NO_INLINE would be that great.
1. littlefs still writes a piece of metadata (12 bytes + padding) to the directory for every sync/close. For non-inline files this is just a pointer to the file data.
2. littlefs rewrites the last block being written to in non-inlined files. So it would still need to rewrite the same amount of data just in a different location.

littlefs (and by necessity any wear-leveling filesystem) is a bit different from a traditional filesystem in that the data is always moving. So it usually works out best if we just keep the amount of data we need to write as small as possible, even if it changes formats several times in its lifetime.

That being said, there's no reason not to try it once we have reproducible benchmarks running.