littlefs-project/littlefs #441: lfs_mount -> -22

marcellangmaier, 2020-06-12:
STM32F4XX board with W25Q16JV 2MB SPI flash

My configuration is:
`
int block_device_read(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, void *buffer, lfs_size_t size)
{
	W25qxx_ReadBlock((uint8_t*)buffer, (block * c->block_size), off, size);
	return 0;
}

int block_device_prog(const struct lfs_config *c, lfs_block_t block, lfs_off_t off, const void *buffer, lfs_size_t size)
{
	W25qxx_WriteBlock((uint8_t*)buffer, (block * c->block_size), off, size);
	return 0;
}

int block_device_erase(const struct lfs_config *c, lfs_block_t block)
{
	W25qxx_EraseBlock(block * c->block_size);
	return 0;
}

int block_device_sync(const struct lfs_config *c)
{
	return 0;
}

const struct lfs_config cfg = {
    .read  =  block_device_read,
    .prog  = block_device_prog,
    .erase = block_device_erase,
    .sync  = block_device_sync,

	.read_size = 256,
	.prog_size = 256,
	.block_size = 4096,
	.block_count = 512,
	.lookahead_size = 64,   // >= blockcount/8
 	.cache_size = 256,     // == read_size256
	.block_cycles = 1000,

	.name_max = LFS_NAME_MAX,
	.file_max = LFS_FILE_MAX,
};
`
What I am executing:
`
	dbg("initialize flash...\r\n");
	if(!W25qxx_Init()) {
		dbg("error...");
		return;
	}

	int err;
	dbg("formatting ffs...\r\n");
	err = lfs_format(&lfs, &cfg);
	if(!err) {
		dbg("mouting ffs...\r\n");
		err = lfs_mount(&lfs, &cfg);
	}
`
What I am getting:
`
initialize flash...
w25qxx Init Begin...
w25qxx ID:0xEF4015
w25qxx Chip: w25q16
w25qxx Page Size: 256 Bytes
w25qxx Page Count: 8192
w25qxx Sector Size: 4096 Bytes
w25qxx Sector Count: 512
w25qxx Block Size: 65536 Bytes
w25qxx Block Count: 32
w25qxx Capacity: 2048 KiloBytes
w25qxx Init Done
formatting ffs...
../Core/Src/lfs.c:3603:trace: lfs_format(0x200000ec, 0x800bb94 {.context=0, .read=0x8005ad7, .prog=0x8005b03, .erase=0x8+++w25qxx ReadBlock:4096, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:1048576, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
w25qxx EraseBlock 0 Begin...
w25qxx EraseBlock done after 186 ms
+++w25qxx ReadBlock:0, Offset:256 ,Read 256 Bytes, begin...
w25qxx ReadPage:1, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx WriteBlock:0, Offset:0 ,Write 256 Bytes, begin...
w25qxx WritePage:0, Offset:0 ,Writes 256 Bytes, begin...
w25qxx WritePage done after 10 ms
---w25qxx WriteBlock Done
+++w25qxx ReadBlock:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:4096, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:1048576, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:0, Offset:256 ,Read 256 Bytes, begin...
w25qxx ReadPage:1, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
w25qxx EraseBlock 4096 Begin...
w25qxx EraseBlock done after 174 ms
+++w25qxx ReadBlock:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:4096, Offset:256 ,Read 256 Bytes, begin...
w25qxx ReadPage:1048577, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx WriteBlock:4096, Offset:0 ,Write 256 Bytes, begin...
w25qxx WritePage:1048576, Offset:0 ,Writes 256 Bytes, begin...
w25qxx WritePage done after 10 ms
---w25qxx WriteBlock Done
+++w25qxx ReadBlock:4096, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:1048576, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
../Core/Src/lfs.c:3664:trace: lfs_format -> 0
                                             mouting ffs...
../Core/Src/lfs.c:3684:trace: lfs_mount(0x200000ec, 0x800bb94 {.context=0, .read=0x8005ad7, .prog=0x8005b03, .erase=0x800+++w25qxx ReadBlock:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:4096, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:1048576, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:0, Offset:256 ,Read 256 Bytes, begin...
w25qxx ReadPage:1, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
+++w25qxx ReadBlock:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage:0, Offset:0 ,Read 256 Bytes, begin...
w25qxx ReadPage done after 0 ms
---w25qxx ReadBlock Done
../Core/Src/lfs.c:3813:trace: lfs_unmount(0x200000ec)
                                                     ../Core/Src/lfs.c:3815:trace: lfs_unmount -> 0
      ../Core/Src/lfs.c:3808:trace: lfs_mount -> -22

`
Anyone an idea whats wrong? Is anyone using littlefs on this W25Q16JV  chip?

e107steved, 2020-06-12:
I Suggest a careful look at your routines which convert from lfs block/offset to absolute memory addresses. And maybe where your "print" routines get data from; some of the values don't look right (without having checked in detail):
```
w25qxx ReadBlock:4096
w25qxx ReadPage:1048576
```

techNiv1, 2020-09-04:
hello @marcellangmaier sir,
How to implement a wear leveling file system for NOR serial flash? is Littlefs better way? if yes how to use littlefs to create a filesystem?
If u could share some piece of code of how u have implemented it, it will be really helpful to me.

gtaska, 2021-02-04:
Probably a bit late for any advice, but @marcellangmaier are you sending a WriteEnable command before writing each PageProgram command (and not just at the start of writing a page if you are writing to the page in multiple SPI transactions)?

I'm working with a similar device and was only writing the first handful of bytes for every page I didn't realize that _every_ PageProgram transaction needed a preceding WriteEnable, regardless of whether you are performing a partial or full page write.

The same goes for the Sector/Block/Chip Erase commands.