littlefs-project/littlefs #446: multiple partitions

giomaca-flo, 2020-07-03:
Hello to everyone,
I'd like to create two partition using littlefs, but I haven't found the way to do this.

There is a way to create more partition using littlefs?

Could someone help me?

Thanks,
Giovanni

e107steved, 2020-07-03:
You can create more than one drive in littleFs - you create and use two separate file systems, each with a separate file definition block (type lfs_t). These may be on different memories, or the same memory.
If you use the same block of memory for both file systems, decide how much memory is to be allocated to each file system. The low-level drivers which are called by littleFs receive a block number and offset _within that file system's memory_, and your drivers need to convert that into a start address within the right part of the memory.

judgeh, 2020-07-15:
similar to what @e107steved said,   I added a "block offset" item to the lfs_configuration so I can generically calculate the correct block offset in the block device functions .. This allowed me to initialize multiple lfs partitions that all use the same block device functions.  my primary partition starts at block_offset=0, while my secondary partition starts at block_offset=1024