littlefs-project/littlefs #449: LittleFS on an Arduino's EEPROM

nobodyinperson, 2020-07-10:
Hi everyone,

Thanks very much for LittleFS!

I'd very much like to use LittleFS on my Arduino (MEGA)'s EEPROM. I'm already using LittleFS on my ESP8266s and for compatibility reasons would really like to have it on my Arduinos as well. 

I couldn't really find anything to start with:

- There are [issues mentioning LittleFS on an EEPROM](https://github.com/ARMmbed/littlefs/issues?q=eeprom), but not quite on an Arduino.
- There are [issues mentioning Arduinos](https://github.com/ARMmbed/littlefs/issues?q=arduino), but don't provide any examples.
- It seems to be possible to port LittleFS to something different than mbedOS (see #382), but no mention of how to do that with an EEPROM on an Arduino.
- The ESP8266 Arduino Framework [provides a LittleFS port](https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS), but that's storing on the ESP8266s Flash memory, not on EEPROM (which it doesn't even have).
- There don't seem to be any [Arduino-related LittleFS-libraries on the PlatformIO Registry](https://platformio.org/lib/search?query=littlefs%2520framework%253Aarduino%2520platform%253Aatmelavr&page=1)

It surely is possible to use LittleFS on an EEPROM, right? It just seems nobody has ever attempted to do it on an Arduino? Or am I just blind?

Any ideas? 

Cheers,

Yann

e107steved, 2020-07-10:
It's perfectly possible to port LittleFS to pretty much any system capable of compiling the code (as you yourself have highlighted, since you're using it on an ESP8266). At the lower level you just make the appropriate calls to littleFS. (If you look at some of the issues, many if not most of the contributors are running littleFs in a variety of environments; not just mbedOS).
At the hardware level its up to you to write drivers which interface between littleFs and your memory; these are relatively simple (close to the sort of thing you might write to test the hardware setup). And hardware is so variable that the lack of examples should come as no surprise. The documentation of what is required is pretty good (extremely good for an open source project, IMO) and there are further explanations in responses to some of the issues. So I suggest you get reading; here, and in the lfs.h file.
Main thing with EEPROM (as with flash) is that it takes a finite time to program, so you have to manage that appropriately. And you can run with a minimum block size of 128 bytes.

geky, 2020-07-17:
@e107steved those are good highlights.

The documentation is one thing I do want to improve, (especially around porting).

---

Two extra points:
1. Look at the `struct lfs_config`, this is where littlefs gets its runtime configuration including block device functions:
   https://github.com/ARMmbed/littlefs/blob/master/lfs.h#L151
2. If your compiler is special you may also need to modify lfs_util.h. If you don't already know that your compiler is special, I would try compiling as is, and then only change things if it's not compiling. If you're desperate, LFS_NO_INTRINSICS will fall back to software-only implementations of intrinsics.
   https://github.com/ARMmbed/littlefs/blob/master/lfs_util.h

---

> It surely is possible to use LittleFS on an EEPROM, right?

How large is the EEPROM?

From my experience with EEPROM it's one of the easiest storage devices to use, but also the least dense. So littlefs should work fine, except if you have less than say ~2KiB, you may run into problems.

e107steved, 2020-07-17:
> From my experience with EEPROM it's one of the easiest storage devices to use,

I actually use FRAM, which is even easier! Basically non-volatile SRAM, so byte at a time access, and no special requirements for writing. But it does cost rather more than EEPROM.

aentinger, 2023-03-23:
[Here's](Works with EEPROM, see [here](Yes it can, I just tried it for myself, see [here](https://github.com/107-systems/107-Arduino-littlefs/blob/main/examples/EEPROM/EEPROM.ino).). :wink: :bow:) an example of littlefs working on EEPROM within Arduino :wink: :bow: .