littlefs-project/littlefs #452: lfs.c:998:error: Corrupted dir pair at {0xc, 0xd}

King-Nick, 2020-07-20:
using latest Littlefs version
my config is :
const struct lfs_config cfg = {
	.context = NULL,
	// block device operations
	.read = my_lfs_bd_read,
	.prog = my_lfs_bd_prog,
	.erase = my_lfs_bd_erase,
	.sync = my_lfs_bd_sync,

	// block device configuration
	.read_size = 64,
	.prog_size = 64,
	.block_size = 4096,  // 4K
	.block_count = 128,  // block_size*block_count = 12K
	.block_cycles = 500,
	.cache_size = 64,
	.lookahead_size = 128,
};

It can work for a while, but there is an error that the directory cannot be found.
It will cause system recognition errors during startup, and automatic reformatting will cause file loss.
use Code：

ret = lfs_file_open(&lfs, &file, fbuf, LFS_O_WRONLY | LFS_O_CREAT | LFS_O_EXCL);
	if (ret >= 0) {
		lfs_file_write(&lfs, &file, (char *)p_usr, sizeof(usr_inf_t));
		ret = lfs_file_close(&lfs, &file);
		if (ret >= 0) {
			errs = 1;
		} else {
			errs = ret;
		}
	} else {
		// file already
		if (ret == LFS_ERR_EXIST) {
			ret = lfs_file_open(&lfs, &file, fbuf, LFS_O_RDWR);
			if (ret >= 0) {
				// size
				len = lfs_file_size(&lfs, &file);
				if (len >= 0) {
					WSLOG_STA("size %d", len);
					if (len == 0) {
						lfs_file_rewind(&lfs, &file);
						lfs_file_write(&lfs, &file, (char *)p_usr, sizeof(usr_inf_t));
					} else {
						p_buf = (char *)malloc(sizeof(usr_inf_t));
						if (p_buf != NULL) {
							lfs_file_rewind(&lfs, &file);
							ret = lfs_file_read(&lfs, &file, p_buf, sizeof(usr_inf_t));
							if (ret >= 0) {
								// same
								if (memcmp((char *)p_usr, p_buf, sizeof(usr_inf_t) - sizeof(uint8_t)-sizeof(friendbean_inf_t)) == 0) {
									WSLOG_STA("info same");
								} else {
									// diff
									WSLOG_STA("info diff");
									lfs_file_rewind(&lfs, &file);
									lfs_file_write(&lfs, &file, (char *)p_usr, sizeof(usr_inf_t));
								}
							} else {
								WSLOG_WARN("read err");
								//force write
								lfs_file_rewind(&lfs, &file);
								lfs_file_write(&lfs, &file, (char *)p_usr, sizeof(usr_inf_t));
							}
							free(p_buf);
						} else {
							WSLOG_WARN("mem err");
							//force write
							lfs_file_rewind(&lfs, &file);
							lfs_file_write(&lfs, &file, (char *)p_usr, sizeof(usr_inf_t));
						}
					}
				}
				ret = lfs_file_close(&lfs, &file);
				if (ret >= 0) {
					errs = 1;
				} else {
					errs = ret;
				}
			}
		} else {
			errs = ret;
		}
	}

e107steved, 2020-07-22:
How confident are you of your hardware, and your low-level drivers? LittleFs is sufficiently mature that "random" errors like yours are fairly rare. Or is there other code which could be corrupting some data somewhere?

Incidentally, that's quite an old version of GCC you're using, although unlikely to be a problem.

King-Nick, 2020-07-23:
It is suspected whether the data address has been tampered with, causing the data to be unrecognizable.
During the test, I found that when one folder is damaged, the other folder can be opened normally.

King-Nick, 2020-07-23:
> 您对自己的硬件和底层驱动程序有多自信？LittleFs非常成熟，因此像您这样的“随机”错误非常少见。还是有其他代码可能破坏某处的某些数据？
> 
> 顺便说一句，虽然您不太可能会遇到问题，但这只是您使用的GCC的较旧版本。

King-Nick, 2020-07-23:
> How confident are you of your hardware, and your low-level drivers? LittleFs is sufficiently mature that "random" errors like yours are fairly rare. Or is there other code which could be corrupting [...]

It is suspected whether the data address has been tampered with, causing the data to be unrecognizable.
During the test, I found that when one folder is damaged, the other folder can be opened normally.

closed 2020-07-23 by King-Nick