littlefs-project/littlefs #456: Is lfs_config need to be persisent?

Roger-Waters, 2020-08-17:
Hello. Is it safe to free memory for `struct lfs_config` after `lfs_mount()` call?

geky, 2020-08-17:
Hi @blomnik, yes lfs_config needs to persist. littlefs depends on it after mount.

The reason for this is that lfs_config is immutable and can be stored in flash this way, rather than copying the config into RAM in the lfs_t struct.

This may change in the future. The configuration is something that I'd like to look at again, and the idea of enabling compile-time configuration has been floated by other developers. But if that happens a persistent lfs_config can still be provided without breaking backwards compatibility.

blomnik, 2020-08-17:
Oh, thank you for that fast repsonse. :)
Does it mean, that littlefs does not modify any field in `struct lfs_config` behind my back? So, this structure can be completly `const`?

Thank you.

geky, 2020-08-18:
Yep! lfs_mount/lfs_format take in a `const*`. The intention is for lfs_config to be stored in flash.

This doesn't do much if the configuration is calculated dynamically, such as in OSs, but it is a nice benefit for devices with very limited memory.