littlefs-project/littlefs #457: How about EEPROM

FreddyBZ, 2020-08-17:
LittleFS seem to be optimized for usage with flash memory. Any issues I need to be aware of if I want to use it on EEPROM?

geky, 2020-08-18:
https://github.com/ARMmbed/littlefs/issues/449 may be related

There's a minimum block size of 128-bytes. However this also impacts the maximum file name limits, so you may consider using 512 byte blocks as a well tested value.

Other than that it should work fine. You can set the block device `erase` function to be a noop and do all the writing work in the block device `prog` function.

You my run into issues if your EEPROM is aggresively tiny (<1KiB?), but littlefs is designed to work even if there are only two erase blocks. So hopefully it won't be an issue.

FreddyBZ, 2021-05-14:
@geky,

It took me till now to actually implement it and I think I've found that your reply above had one error. I cannot set the erase function to noop. Looks like LittleFS relies on having FF's in erased blocks (at least at some locations). For now I just write FF's over the whole block what it needs to be erased, but I don't like it... Did I misunderstand something?

Freddy

geky, 2021-06-12:
Hmmm, I'm not sure why that would be. LittleFS relies on this behavior for SD/eMMC (writing FFs would be very expensive).

Perhaps try testing locally with a configuration that simulates EEPROM? There may be something else going wrong.

- https://github.com/littlefs-project/littlefs/blob/1863dc7883d82bd6ca79faa164b65341064d1c16/.github/workflows/test.yml#L100-L105
- https://github.com/littlefs-project/littlefs/blob/1863dc7883d82bd6ca79faa164b65341064d1c16/tests/test_badblocks.toml#L13

(It may be easier to write a custom test case configured with LFS_TESTBD_BADBLOCK_ERASENOOP than getting the full test suite to pass)

pavel-kirienko, 2021-07-01:
>I cannot set the erase function to noop. Looks like LittleFS relies on having FF's in erased blocks (at least at some locations). For now I just write FF's over the whole block what it needs to be erased, but I don't like it... Did I misunderstand something?

FWIW, I was able to make LittleFS function correctly with EEPROM following geky's suggestion above: my sync() and erase() are no-ops and there appear to be no issues. Perhaps your memory driver is defective?

FreddyBZ, 2021-07-01:
@geky @pavel-kirienko I don't have time to play with it, but looks like after erase of a block some block number field in the block is assumed to be 0xffffffff - which won't happen if erase is implemented as a noop. Could be that this was fixed in the meantime - my original version was v2.1.4, now I've updated to v2.4.1.

pavel-kirienko, 2021-07-01:
>Could be that this was fixed in the meantime - my original version was v2.1.4, now I've updated to v2.4.1.

That could be the case, hypothetically. I only have experience with v2.4.1.

aentinger, 2023-03-23:
Works with EEPROM, see [here](Yes it can, I just tried it for myself, see [here](https://github.com/107-systems/107-Arduino-littlefs/blob/main/examples/EEPROM/EEPROM.ino).). :wink: :bow: