littlefs-project/littlefs #458: Is lfs_file_open implementation limited to one open file at a time?

FreddyBZ, 2020-08-18:
Looks like lfs_file_open is using a static lfs_file_config struct. If I understand correctly this struct is per open file, so only one file can be open at one time with lfs_file_open. I know that it can be overcome by using lfs_file_opencfg - I just want to be sure I understand it correctly.

Thanks,

Freddy

geky, 2020-08-19:
The lfs_file_config struct is similar to the lfs_config config struct. It needs to be persistent, but it's readonly and can be placed in flash. This means it's safe to reference across multiple file structs since multiple open files don't modify the underlying lfs_config struct.

It's like a bunch of people look at a painting, it's fine. But if you all try to paint on the same canvas it's going to get messy.

I'm not the best at metaphors.

FreddyBZ, 2020-08-20:
I see. The problem seem to be only when using lfs_file_opencfg with a user provided buffer in the lfs_file_config - which I assume still needs to be per file. In the lfs_file_open function the buffer pointer is null and and a buffer per file is allocated dynamically.

Thanks.

xennex22, 2020-12-19:
I think the current behavior is acceptable as far as cache buffers goes and the use of the static, shared, config.

If you are using `lfs_open()` with lfs doing the memory allocation (`LFS_NO_MALLOC` not defined) then the static, zeroed, config from `lfs_file_rawopen` is passed to `lfs_file_rawopencfg` which then allocates a buffer.  So multiple files can be opened.

You can't use 'lfs_open' with your own memory allocation (`LFS_NO_MALLOC` defined) because it has no way of allocating a buffer.

So using 'lfs_opencfg' with your own memory allocation (`LFS_NO_MALLOC` defined) then you should have allocated a config with a pointer to an allocated buffer.  The only slight issue is that when it comes to cleanup time `lfs_file_t` has a `const struct lfs_file_config` pointer so you have to cast to be able to free the config.