littlefs-project/littlefs #462: Directory data written to 0x00000100 and following (All page 0) by lfs_file_open but never read from there by lfs_dir_read.

bastetfurry, 2020-08-23:
At the title says, i create a zero byte file on an Winbond W25Q32 flash connected trough SPI on a STM32F446RE and when i try to read the directory it won't show up.
While tracing it out i saw that on file creation littlefs writes, i presume, directory data to 0x0100 and onward.
If i try to read back the directory i only get the first entry as the directory reader never tries to access the other blocks in 0x0100 but starts first at 0x0000 and then continues at 0x1000, block 1, never reading from block 0 again.
If i try to touch the same file again the file routines seem to see that the file is already in the directory tough.

I have no clue anymore what could be wrong, maybe one of the devs see whats going on and if the bug is on my side or not. :(

The routine that creates the file:
```
void doTouch(char *filename)
{
	lfs_t lfs = SPIFlash_GetLfsStruct(activedrive);

	lfs_file_t fp;

	int error = lfs_file_open(&lfs, &fp, filename, LFS_O_WRONLY | LFS_O_CREAT);
	if(error < 0)
	{
		printf("Error while touching file: %d\n",error);
		return;
	}
#ifdef SHELL_PRINT_DEBUG
		printf("[Shell][Debug] lfs_file_open error code:%d\n",error)
#endif
	lfs_file_close(&lfs, &fp);
}
```

The routine that reads the directory:
```
void doDir()
{
	int done = 0, error;
	lfs_t lfs = SPIFlash_GetLfsStruct(activedrive);
	struct lfs_config cfg = SPIFlash_GetLfsConfig(activedrive);

	lfs_dir_t lfs_dir;
	struct lfs_info entryinfo;
	lfs_ssize_t lfs_ssize = lfs_fs_size(&lfs);

	lfs_dir_open(&lfs, &lfs_dir, "/");

	while(done == 0)
	{
		error = lfs_dir_read(&lfs, &lfs_dir, &entryinfo);

#ifdef SHELL_PRINT_DEBUG
		printf("[Shell][Debug] lfs_dir_read error code:%d\n",error)
#endif
		if(error <= 0)
			done = 1;

		printf("%s %-10d %s\n"
				,(entryinfo.type==LFS_TYPE_REG)?"PRG":"DIR"
				,entryinfo.size
				,entryinfo.name
				);
	}

	printf("Total size: %-10d Used: %-10d\n",cfg.block_size*cfg.block_count,lfs_ssize);

	lfs_dir_close(&lfs, &lfs_dir);
}
```

Maybe my trace including debug info is helpful here:
```
>touch asdfg
../Core/Src/lfs.c:2540:trace: lfs_file_open(0x2001fd4c, 0x2001fcf8, "asdfg", 102)
../Core/Src/lfs.c:2389:trace: lfs_file_opencfg(0x2001fd4c, 0x2001fcf8, "asdfg", 102, 0x804410c {.buffer=0, .attrs=0, .attr_count=0})
[... 24 lines cut ...]
[SPIFlash][Debug] Reading 256 bytes from block 0x00000001, offset 0x00000000 (Address: 0x00001000) from drive 0:02 00 00 00 f0 0f ff f7 6c 69 74 74 6c 65 66 73 2f e0 00 10 00 00 02 00 00 10 00 00 00 04 00 00 ff 00 00 00 ff ff ff 7f fe 03 00 00 70 1f fc c8 e9 ec 5b bc ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[SPIFlash][Debug] Reading 256 bytes from block 0x00000000, offset 0x00000000 (Address: 0x00000000) from drive 0:01 00 00 00 f0 0f ff f7 6c 69 74 74 6c 65 66 73 2f e0 00 10 00 00 02 00 00 10 00 00 00 04 00 00 ff 00 00 00 ff ff ff 7f fe 03 00 00 70 1f fc c8 8d d9 89 fa ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[SPIFlash][Debug] Reading 256 bytes from block 0x00000001, offset 0x00000000 (Address: 0x00001000) from drive 0:02 00 00 00 f0 0f ff f7 6c 69 74 74 6c 65 66 73 2f e0 00 10 00 00 02 00 00 10 00 00 00 04 00 00 ff 00 00 00 ff ff ff 7f fe 03 00 00 70 1f fc c8 e9 ec 5b bc ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[SPIFlash][Debug] Reading 256 bytes from block 0x00000001, offset 0x00000100 (Address: 0x00001100) from drive 0:10 1f f8 d0 40 00 00 03 30 30 31 20 00 00 03 70 1f f8 ed 60 87 1e 46 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[SPIFlash][Debug] Reading 256 bytes from block 0x00000001, offset 0x00000200 (Address: 0x00001200) from drive 0:ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
../Core/Src/lfs.c:2055:trace: lfs_dir_open -> 0
../Core/Src/lfs.c:2075:trace: lfs_dir_read(0x2001fd44, 0x2001fcc4, 0x2001fbbc)
../Core/Src/lfs.c:2083:trace: lfs_dir_read -> 1
DIR 0          .
../Core/Src/lfs.c:2075:trace: lfs_dir_read(0x2001fd44, 0x2001fcc4, 0x2001fbbc)
../Core/Src/lfs.c:2089:trace: lfs_dir_read -> 1
DIR 0          ..
../Core/Src/lfs.c:2075:trace: lfs_dir_read(0x2001fd44, 0x2001fcc4, 0x2001fbbc)
[SPIFlash][Debug] Reading 256 bytes from block 0x00000001, offset 0x00000100 (Address: 0x00001100) from drive 0:10 1f f8 d0 40 00 00 03 30 30 31 20 00 00 03 70 1f f8 ed 60 87 1e 46 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[SPIFlash][Debug] Reading 256 bytes from block 0x00000001, offset 0x00000000 (Address: 0x00001000) from drive 0:02 00 00 00 f0 0f ff f7 6c 69 74 74 6c 65 66 73 2f e0 00 10 00 00 02 00 00 10 00 00 00 04 00 00 ff 00 00 00 ff ff ff 7f fe 03 00 00 70 1f fc c8 e9 ec 5b bc ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff [... cut ...] f ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
../Core/Src/lfs.c:2122:trace: lfs_dir_read -> 1
PRG 0          001
../Core/Src/lfs.c:2075:trace: lfs_dir_read(0x2001fd44, 0x2001fcc4, 0x2001fbbc)
../Core/Src/lfs.c:2096:trace: lfs_dir_read -> 0
DIR 0
Total size: 4194304    Used: 2
../Core/Src/lfs.c:2060:trace: lfs_dir_close(0x2001fd44, 0x2001fcc4)
../Core/Src/lfs.c:2069:trace: lfs_dir_close -> 0
```

bastetfurry, 2020-08-23:
Update: If i create 17 files that way all suddenly show up in the directory listing.

geky, 2020-12-08:
Hi @bastetfurry, thanks for opening an issue, sorry for the late response.

Hmm, the reason littlefs doesn't read the contents of block 0 is because it determined that the contents of block 1 are more recent. littlefs uses a revision count (the first 32-bit little-endian integer in the block) to determine which block is the most recent. But littlefs shouldn't have written the new file to block 0 without incrementing the revision count (0 is fine, the revision count is allowed to roll over and 0xffffffff is usually the initial state of flash).

The 17 file switch is extra weird, because once a revision count increments, the previous block's contents should never be used again.

My guess is something is change the on-disk contents when littlefs doesn't expect it to, but it's hard to say.

Is this system multithreaded by any chance?

bastetfurry, 2020-12-12:
Nope, no multiplexing here. The goal is some 70s-retro-style console with a seven segment display array and console access trough serial with cartridges based on SPI Flash for storage, no need for multitasking here. ;)

If it helps attached is my whole bridge, spiflash.c and .h, from littlefs to the flash chips, maybe i am just stupid and missing something. -.-
[spiflash_c_h.zip](https://github.com/littlefs-project/littlefs/files/5683081/spiflash_c_h.zip)