littlefs-project/littlefs #464: @Johnxjj , You are correct i had removed lfs_file_opencfg()  and used only lfs_file_open()  and it works.

vinaykskl1, 2020-08-30:
Hi @Johnxjj ,

I have the similiar issue with lfs_format. I have used the configuration below(flash size is 64Mbyte) --
#define DATA_STORE_READ_SIZE        512
#define DATA_STORE_WRITE_SIZE       512 
#define DATA_STORE_BLOCK_SIZE       262144UL            // (256K)
#define DATA_STORE_BLOCK_COUNT      256 
#define DATA_STORE_CACHE_SIZE       2048  
#define DATA_STORE_LOOKAHEAD_SIZE   512
#define DATA_STORE_BLOCK_CYCLES     500

static uint8_t lfs_readBuff[DATA_STORE_READ_SIZE];
static uint8_t lfs_writeBuff[DATA_STORE_WRITE_SIZE];
static uint8_t __attribute__((__aligned__(8))) lfs_lookaheadBuff[DATA_STORE_LOOKAHEAD_SIZE];

It gives the error LFS_ERR_CORRUPT at the point of crc checks under lfs_dir_commitcrc() function--

    // successful commit, check checksums to make sure
    lfs_off_t off = commit->begin;
    lfs_off_t noff = off1 + sizeof(uint32_t);
    while (off < end) {
        uint32_t crc = 0xffffffff;
        for (lfs_off_t i = off; i < noff+sizeof(uint32_t); i++) {
            // check against written crc, may catch blocks that
            // become readonly and match our commit size exactly
            if (i == off1 && crc != crc1) {
                return LFS_ERR_CORRUPT;
            }

could you please get me resolution for the same.

vinaykskl1, 2020-09-01:
If anyone has solution to this, please let me know.
the code returns the error LFS_ERR_CORRUPT in crc verification.

static int lfs_dir_commitcrc(lfs_t *lfs, struct lfs_commit *commit) {
    const lfs_off_t off1 = commit->off;
    const uint32_t crc1 = commit->crc;
    // align to program units
   .
   .
   .
    // successful commit, check checksums to make sure
    lfs_off_t off = commit->begin;
    lfs_off_t noff = off1 + sizeof(uint32_t);
    while (off < end) {
        uint32_t crc = 0xffffffff;
        for (lfs_off_t i = off; i < noff+sizeof(uint32_t); i++) {
            // check against written crc, may catch blocks that
            // become readonly and match our commit size exactly

            if (i == off1 && crc != crc1) {
                return LFS_ERR_CORRUPT;
            }

barani143, 2020-09-07:
Hi vinayksk11,
Look like your block device driver having some problem. Check the flash write function.

zhabl, 2020-10-14:
Hi, 
lfs_bd_read() would bypass catche, and this cause lfs_format() error in my project,  because we use DMA in device driver, and some
memeroy bank can not be accessed by DMA,.   May be this help.

chandrasekharmorisetti, 2023-03-01:
Is there a resolution to this issue reported by @vinaykskl1 
If so, please share the same.

chandrasekharmorisetti, 2023-03-03:
Yes, this problem comes when your block device is locked for writing. You need to unlock the device for writing and then attempt to call FS related API's. We did face this issue and by unlocking the BD (serial flash) no errors related to bad block appeared

It is always good to write a simple write and read functions on to your block device (BD) independently and check if write and read are working fine. After that integrating them into FS will help and also avoid issues mentioned in this issue.

XieyangLy, 2023-11-03:
> If anyone has solution to this, please let me know. the code returns the error LFS_ERR_CORRUPT in crc verification.
>
> static int lfs_dir_commitcrc(lfs_t *lfs, struct lfs_commit *commit) { const [...]

hi,i meet this problem too. have you solve it?
and,i do some code to check the the wirite read function. mostly it can be work.
but the lsf will return LFS_ERR_CORRUPT every time

XieyangLy, 2023-11-04:
i found a way to solve this problem:
just erase full chip,before you lfs_format the fs。

ps : it mentioned in #71 .
> hi,i meet this problem too. have you solve it? and,i do some code to check the the wirite read function. mostly it can be work. but the lsf will return LFS_ERR_CORRUPT every time