littlefs-project/littlefs #466: Max file size in nand flash

Dothrakki, 2020-09-02:
Hi

I am using the littlefs with winbond nand flash, w25n01gv,  and all the basic functions are working, even though i have ran into some performance issues.

i am facing some issues while trying increase the file size more than the block size. The the nand flash internal block size is 128k and which is the smallest erasable portion.The data is getting corrupted  if i try to increase the lfs blocksize. 
i am using lfs release 2.14.

#define LFS_READ_SIZE						4096
#define LFS_PROG_SIZE						4096
#define LFS_BLOCK_SIZE					(1024*128)
#define LFS_BLOCK_COUNT					1024
#define LFS_CACHE_SIZE					4096
#define LFS_LOOKAHEAD_SIZE			2048
#define LFS_BLOCK_CYCLES				500

gofortime, 2020-09-07:
I am using nand, and it's fine to use file size more than bock size (in my case 256KB).

Dothrakki, 2020-09-07:
Thank you for the kind response.

can u please share your configuration? 
i am not sure why my data is getting corrupted.

geky, 2020-09-29:
">block size" is when files stop being one block and start using the CTZ skip-list. Maybe try defining `-DLFS_NO_INTRINSICS=1`?