littlefs-project/littlefs #467: lfs_dir_commit() hits LFS_ASSERT()

ptahpeteh, 2020-09-08:
Hi,

The call to lfs_mkdir() results in LFS_ASSERT being called in the lfs_dir_commit(), line 1746:

```
} else if (lfs_tag_type3(attrs[i].tag) == LFS_TYPE_DELETE) {
    LFS_ASSERT(dir->count > 0);
    dir->count -= 1;
    hasdelete = true;
}
```
The trace:
```
lfs.c:3795:debug: Found pending gstate 0x4ff00000000006c7000006c6
lfs.c:4122:debug: Fixing move {0x6c7, 0x6c6} 0x0
```
The stack trace was as follows:
```
lfs_fs_demove()
lfs_fs_forceconsistency()
```
What can be the reason for this? Is there any other way of getting out of this issue other than reformat?
I don't understand why ASSERT is used here... Is it possible to remove the ASSERT and change it to this:
```
if (dir->count > 0) {
    dir->count -= 1;
}

hasdelete = true;
```

geky, 2020-09-29:
Hi @ptahpeteh, thanks for creating an issue. Unfortunately I probably won't be able to dig into what's going on for a while so I can only offer some pointers. This is in a particularly complicated state of littlefs.

> ```
> lfs.c:3795:debug: Found pending gstate 0x4ff00000000006c7000006c6
> lfs.c:4122:debug: Fixing move {0x6c7, 0x6c6} 0x0
> ```

littlefs found a move in the metadata-pair contained in the two blocks 0x6c7 and 0x6c6. At some point power was lost in the middle of a move (lfs_rename) operation.

Because littlefs contains a traditional filesystem tree (great for performance), when you move between different directories it can't actually do that atomically. Instead it will create the file reference in the destination while atomically updating a small bit of "distributed global state" (gstate) to indicate that a move is occurring. Then delete the source while atomically updating the gstate to indicate the move has been taken care of.

If power is lost, littlefs should see the "pending gstate" and fix the move by deleting the file that was moved. That's what should have happened.

---

The assert triggers because, while trying to delete the file that was moved, littlefs found no file. This shouldn't happen.

It may be more correct to return LFS_ERR_CORRUPT in this situation, as the filesystem has become corrupted, but I don't know if continuing as though nothing happened is ok. If implemented as suggested, a power loss could cause a lfs_rename to result in two files. This could cause problems for dependent applications that expect lfs_rename to be atomic.

Unfortunately I don't know why this could be happening. Keep in mind the update to remove the old file atomically clears the move gstate, and runs prior to the first write to the filesystem. It shouldn't be possible for littlefs to remove the file without also clearing the gstate and likely something worse has gone wrong in the internals of littlefs's logic.

ptahpeteh, 2020-09-30:
Hi @geky. Thank you for your response.

I think a better idea would be to allow client code (application) to decide what to do with such errors. Instead of calling assert it would call the fault handler provided by the application. And the application could for example reformat the flash and reboot the device. This could bring the device back to life. What do you think about it? Because in my case file system corruption resulted in a blocked device. Assert stopped the program and then it was restarted by watchdog. After restart again assert, and so on ad infinitum.

And I have another question. As I was reading through lfs code I found  this `LFS_ASSERT()`:
```
int err = lfs->cfg->prog(lfs->cfg, pcache->block, pcache->off, pcache->buffer, diff);
LFS_ASSERT(err <= 0);
```
This is line 146 of lfs.c, `lfs_bd_flush()` function. Why assert here instead of returning an error?

geky, 2020-10-06:
You're right, this should be an error :+1: Especially since it can result from a corrupted filesystem image.

The assert itself is actually in the `lfs_dir_commit` function, which shouldn't be called with a delete command unless there is a valid ID to delete. So this is actually a failure that comes from a different part of littlefs's logic.

What should be done is the `lfs_dir_demove` function should check the directory size when it fetches the directory pair, and return an error there:
https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L4126

---

> This is line 146 of lfs.c, `lfs_bd_flush()` function. Why assert here instead of returning an error?

Ah, this is to protect against block device operations returning positive error codes. They should only return `0` for success, `LFS_ERR_CORRUPT` to indicate a block is not valid, or a negative error code on failure, which is propagated to the user.

littlefs doesn't handle partial writes, so a returned size would always need to be the same as the requested size. The option was either `int err = min(0, err)` or assert on positive error codes.

ptahpeteh, 2020-11-18:
Thanks for your comment. What about the other two calls to `lfs_dir_commit` with `LFS_TYPE_DELETE`, one in `lfs_rename()` and another in `lfs_remove()`? Should there also be the check for the directory size? 

I was going through all the LFS_ASSERTs in the code and I don't know what to do with the following ones. They look like they could be changed into returning error code (they assert on data read from flash) but I'm not sure:

https://github.com/littlefs-project/littlefs/blob/master/lfs.c#L142 (lfs_bd_flush)
https://github.com/littlefs-project/littlefs/blob/master/lfs.c#L191 (lfs_bd_prog)
https://github.com/littlefs-project/littlefs/blob/master/lfs.c#L221 (lfs_bd_prog)
https://github.com/littlefs-project/littlefs/blob/master/lfs.c#L233 (lfs_bd_erase)
https://github.com/littlefs-project/littlefs/blob/master/lfs.c#L1667 (lfs_dir_compact)
https://github.com/littlefs-project/littlefs/blob/master/lfs.c#L1840 (lfs_dir_commit)

geky, 2020-12-18:
Sorry for the late response

> one in lfs_rename() and another in lfs_remove()? Should there also be the check for the directory size?

Ah, so these actually look up a file id to remove, which errors with NOENT if the directory is empty:
[https://github.com/littlefs-project/littlefs/blob/1a59954ec64ca168828a15242cc6de94ac75f9d1/lfs.c#L3182 ](https://github.com/littlefs-project/littlefs/blob/1a59954ec64ca168828a15242cc6de94ac75f9d1/lfs.c#L3182)

> I was going through all the LFS_ASSERTs in the code and I don't know what to do with the following ones. They look like they could be changed into returning error code (they assert on data read from flash) but I'm not sure:

These look like they are mostly in writing functions (prog/erase/flush/compact/commit). The conditions being asserted shouldn't be dependent on disk data, if they are that is a mistake.