littlefs-project/littlefs #468: Superblock 0x0 has become unwritable

techNiv1, 2020-09-11:
Hello @geky sir,

I am using LittleFS for NOR flash which is W25Q64. I had ported littlefs library successfully for my controller with all 4 base functions like localread,localProg,localErase and localSync.
while first time mount i got error given below

`custom/lfs.c:1170:error: Corrupted dir pair at {0x0, 0x1}
custom/lfs.c:4418:trace: lfs_unmount(f03d4558)
custom/lfs.c:4420:trace: lfs_unmount -> 0
custom/lfs.c:4411:trace: lfs_mount -> -84
` 
In your comments from the previous few threads i got to know that it is ok to get this error for very first time mount and then you should have to call format function after first mount.
Now I am calling littlefs format function after first mount then i am getting error like

`custom/lfs.c:1970:debug: Bad block at 0x0
custom/lfs.c:1977:warn: Superblock 0x0 has become unwritable
custom/lfs.c:4248:trace: lfs_format -> -28
`
From last line i got to know that it is  `No space left on device`  error.

So i just want to know where am I doing wrong? or where is the issue in my code?
can you please provide me some pointers for this so that i can get out of this problem.
is there any sample codes available to port littlefs library to any microcontroller?

eduards66, 2020-10-07:
LFS_VERSION 0x00020002 :   I got the same error for GD25Q16C, but I can read and write to any address of this chip via QSPI interface:
InitApplication: call to FS_Init() ...
FS_Format: Formatting ...../littlefs/lfs.c:3603:trace: lfs_format(0x200000c4, 0x2000013c {.context=0x200008c0, .read=0x800916d, .prog=0x80091d1, .erase=0x8009231, .sync=0x8009299, .read_size=256, .prog_size=256, .block_size=4096, .block_count=512, .block_cycles=500, .cache_size=256, .lookahead_size=64, .read_buffer=0x20000184, .prog_buffer=0x20000284, .lookahead_buffer=0x20000384, .name_max=32, .file_max=524288, .attr_max=1022})
                                                                        ../littlefs/lfs.c:1685:debug: Bad block at 0x0
                                                                                                                      ../littlefs/lfs.c:1691:warn: Superblock 0x0 has become unwritable
   ../littlefs/lfs.c:3664:trace: lfs_format -> -28
                                                   FAILED!

eduards66, 2020-10-07:
[image]
Please, look on the attached screenshot.  It is exact place where error have place.
Regards,
Eduard.

eduards66, 2020-10-09:
Try to call lfs_bd_flush(..., validate) with validate=true. It will help identify if the problem is related to SPI/QSPI driver.

Vokonn, 2022-02-28:
Hi @techNiv1 

I am using W25N01 with dsPIC33. I have the same issue.  Did you solve the issue?  Would you mind sharing your solution?

littlefs/lfs.c:5009:trace: lfs_mount(1db2, ffffffffffffccc0 {.context=0, .read=3ae, .prog=3b2, .erase=3b6, .sync=3ba, .read_size=1, .prog_size=1, .block_size=131072, .block_count=1024, .block_cycles=512, .cache_size=2048, .lookahead_size=32, .read_buffer=1e6e, .prog_buffer=266e, .lookahead_buffer=3ef0, .name_max=0, .file_max=0, .attr_max=0}). 
littlefs/lfs.c:1084:error: Corrupted dir pair at {0x0, 0x1}.
littlefs/lfs.c:5013:trace: lfs_mount -> -84.

littlefs/lfs.c:4979:trace: lfs_format(1db2, ffffffffffffccc0 {.context=0, .read=3ae, .prog=3b2, .erase=3b6, .sync=3ba, .read_size=1, .prog_size=1, .block_size=131072, .block_count=1024, .block_cycles=512, .cache_size=2048, .lookahead_size=32, .read_buffer=1e6e, .prog_buffer=266e, .lookahead_buffer=3ef0, .name_max=0, .file_max=0, .attr_max=0}). 
littlefs/lfs.c:1806:debug: Bad block at 0x0.
littlefs/lfs.c:1812:warn: Superblock 0x0 has become unwritable.lfs_dir_compact: -28
littlefs/lfs.c:4983:trace: lfs_format -> -28.

ferremarco, 2022-03-15:
I have the same problem with MT29 Micron.
Do you have any update  ?

isoment01, 2022-06-07:
Having Same problem with MX25R8035F.

GiulioDallaVecchia, 2023-04-20:
Having same problem with Windbond Flash

oldmud0, 2023-06-15:
From #71, it seems like LittleFS does not check first if the flash is fully erased before formatting. I had the same error, erased my flash chip, and this fixed the issue.

marcosgabrielserrano, 2024-11-16:
@ferremarco did you ever find a solution? I am running into same problem on MT29 Micron.

pawelzwronek, 2026-05-24:
With a help from claude I manage to fix **Superblock 0x0 has become unwritable** issune on my sd card. My issue raised when I was porting LittleFS to latest v2.11 into my ncs v2.1 (from ncs v3.3). 

Root cause: CONFIG_LITTLE_ENDIAN does not exist in NCS v2.1.2
The updated zephyr_lfs_config.h replaced the endianness detection from:
// ORIGINAL — uses GCC compiler macros, always defined
```c
#if ... __BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__ ...
    return a;  // LE → no-op
```
to:
```c
// AFTER UPDATE — requires a Kconfig symbol
#if defined(CONFIG_LITTLE_ENDIAN)  // ← DOES NOT EXIST in v2.1.2!
    return a;
#elif !defined(LFS_NO_INTRINSICS)
    return __builtin_bswap32(a);   // ← LE machine + bswap = WRONG!
```

Since CONFIG_LITTLE_ENDIAN is absent from both [.config](vscode-file://vscode-app/c:/Users/p/AppData/Local/Programs/Microsoft%20VS%20Code/b6a47e94e3/resources/app/out/vs/code/electron-browser/workbench/workbench.html) and arch/Kconfig in NCS v2.1.2 (it was only introduced in v3.3.0), the functions lfs_tole32/lfs_fromle32 fall through to __builtin_bswap32 on a little-endian machine. This causes the CRC value written to disk to be byte-swapped → the CRC residue check (if (crc != 0)) always returns LFS_ERR_CORRUPT → "Bad block" → "Superblock unwritable".