littlefs-project/littlefs #469: How sector numbers are calculated

Newfatuser, 2020-09-11:
I'm using littlefs for MT25QL128 16MB flash, I've linked littlefs read/ write/ erase functions to QSPI drivers. 
while formatting the flash, I see superblock is written in sector 0, 
// block device configuration
    .read_size = 256,//16,
    .prog_size = 256,//16,
    .block_size = 4096,
    .block_count = 4096,//128,
    .cache_size = 256,//16,
    .lookahead_size = 256,//16,
    .block_cycles = 500,

then  I've created one file, in file open function I see only ID is assigned to file object.
writing 36 bytes data into this file in iterative loop of 100 times. sequence is,
file open
file write
file close
it calls 100 times, what I see is after every 16th iteration it erases sector it erases sector 0 and 1. 
As superblock is written into sector 0 why it is erased while writing file data?
again, I'm not able to understand how the sector numbers are calculated and allocated to file, in file object I see block field is assigned with a long number which doesn't fit into external flash memory address range. please help me understand this sector number calculations.

geky, 2020-09-29:
Hi @Newfatuser, it's a bit complicated. The implementation and reasoning is explained a bit in the [DESIGN.md](https://github.com/ARMmbed/littlefs/blob/master/DESIGN.md).

One thing to keep in mind is that littlefs is a copy-on-write (cow) filesystem. This means files don't remain in the same block, they are moved around as you write to them. This is necessary to provide wear-leveling.

In order to track the file locations as we cow files, this means we need to store the current block somewhere. We end up storing the block address + size as "metadata" in a pair of blocks that doubles as the directory. This "metadata-pair" is updated every time we write to the file, however these are actually implemented as logs, and there is a bunch of logic to move them if they undergo too many writes (based on the block_cycles config option).

There is an extra optimization for small filesystems where the root directory + superblocks share the first two blocks until they have undergone too many writes. This is why you see sector 0 being erased every 16th iteration, enough data was written to the metadata block to trigger a metadata "compact" operation as a part of the metadata logging implementation. The data is actually being copied to block 1 before block 0 is erased. After another 16ish iterations the metadata should be moved back to block 0 and block 1 will be erased.

So it would be more accurate to say the superblock exists on both sector 0 and 1 (these are hardcoded values).

Compared to a full 4KiB sectory, the amount of metadata we need to store in the superblock is very small (roughly 64 bytes), so combining the superblock+root directory is a big benefit for tiny disks, such as 16-block filesystems stored in a device's internal flash. Though this approach does cause problems for automatic filesystem discovery...

---

Normally the first block of the skip-list that backs a file can be found here, in `file.ctz.head`:
https://github.com/ARMmbed/littlefs/blob/master/lfs.h#L336

But seeing as your 4KiB blocks get compacted in only 16-writes, I'm guessing the files are being "inlined". This is another optimization where if the file is small enough and fits in RAM, littlefs will place the entire file in the metadata-pair rather than creating a linked-list.

I'm guessing the "long number" is `(uint32_t)-2`, which is an internal value littlefs uses to track files that are inlined. The two blocks that contain a files metadata are stored here, in `file.m.pair`, though keep in mind a metadata-pair may contain metadata for many files:
https://github.com/ARMmbed/littlefs/blob/master/lfs.h#L336

Newfatuser, 2020-09-29:
Thanks gkey for detailed answer. debugging the littlefs now I've somewhat understanding of the code. currently for writing 4Kb file its taking time between 22mSec to 98mSec, and as the file system groes execution time also increases. I'm trying to reduce execution time for f_write and f_sync function. So, I've tried following options:
1. erased whole flash @powerup time and excluded lfs_bd_erase(lfs, nblock); function call in lfs_ctz_extend and lfs_file_relocate - in this case I was not able to use unused marked memory which is freed after relocation or extension.
2. erased whole flash @powerup time and in lfs_ctz_extend and lfs_file_relocate functions read back 4 bytes from this sector number calculated in nblock if 1st 4 bytes are 0xFF then excluded lfs_bd_erase(lfs, nblock); function call  as 0xFF means it is already erased.
do you think there could be other option which can reduce the execution time.