littlefs-project/littlefs #472: lfs_ctz_extend gives LFS_ERR_CORRUPT from lfs_bd_read

Newfatuser, 2020-10-14:
Hello,
I'm using Littlefs ver #define LFS_VERSION 0x00020002
Nor Flash configurations are 
`// configuration of the filesystem is provided by this struct
const struct lfs_config cfg = {
    // block device operations
    .read  = fs_flash_read,
    .prog  = fs_flash_prog,
    .erase = fs_flash_erase,
    .sync  = fs_flash_sync,

	// block device configuration
	    .read_size = 256,//16,
	    .prog_size = 256,//16,
	    .block_size = 4096,
	    .block_count = 4096,//4096,//128,

	.cache_size = 256,//16,
	.lookahead_size = 256,//8,//256,//16,

    .block_cycles = 1000,//9/30/20: changed for testing 500
};` 
I'm able to create files and write data into it but after some time I'm getting error in function static int lfs_ctz_extend at particular following read function, when I debug it I see **nhead** becomes a huge number(4294967295) which is out of range as a block number limit is 4096. 
`    if (i != skips-1) {
                    err = lfs_bd_read(lfs,
                            NULL, rcache, sizeof(nhead),
							nhead, 4*i, &nhead, sizeof(nhead));
                    nhead = lfs_fromle32(nhead);
                    if (err) {
                        return err;
                    }
                }
`
in the same function there is bd_prog function wherein we're passing **nblock** as a parameter. do we need to pass **nblock** to above bd_read in place of **nhead**? is it a bug? OR Am I missing something?

`               err = lfs_bd_prog(lfs, pcache, rcache, true,
                        nblock, 4*i, &nhead, 4);
                nhead = lfs_fromle32(nhead);
                if (err) {
                    if (err == LFS_ERR_CORRUPT) {
                        goto relocate;
                    }
`

geky, 2020-10-15:
Hmm, 4294967295 == 0xffffffff, so that is suspicious. Is it possible the block device driver isn't writing to the right location? 0xffffffff is the erase state for NOR flash after an erase.

0xffffffff is also used as a placeholder for some block addresses if an open-file's parent directory is removed, but I don't think that's what is happening here...

---

The reason we have separate `nblock` and `nhead` variables is because at that point `nblock` points to the newly allocated block, while we use `nhead` to traverse the CTZ skip-list to populate the new block.

The CTZ skip-list looks something like this:
```
.-------.  .-------.  .-------.  .-------.  .-------.  .-------.
| blk 0 |<-| blk 1 |<-| blk 2 |<-| blk 3 |<-| blk 4 |<-| blk 5 |<-
|       |<-|       |--|       |<-|       |--|       |  |       |
|       |<-|       |--|       |--|       |--|       |  |       |
'-------'  '-------'  '-------'  '-------'  '-------'  '-------'
```

If we want to add a new block "blk 6", we need to find 2 skips (ctz(6)+1 = 2)
```
.-------.  .-------.  .-------.  .-------.  .-------.  .-------.  .-------.
| blk 0 |<-| blk 1 |<-| blk 2 |<-| blk 3 |<-| blk 4 |<-| blk 5 |<-| blk 6 |<-
|       |<-|       |--|       |<-|       |--|       |<-|       |--|       |
|       |<-|       |--|       |--|       |--|       |  |       |  |       |
'-------'  '-------'  '-------'  '-------'  '-------'  '-------'  '-------'
```

We can do this by traversing the skip-list using skips in successive positions, which is what we do in the loop in lfs_ctz_extend. We use `nhead` to track the next skip entry, though it could have a better name...

Newfatuser, 2020-10-26:
Hello,
Thanks for your support and detailed description, yaa it was issue in my application code. Now I’m able to write 16Mb flash successfully.
Same code I’ve modified for 64MB flash. I’m writing a file with 4kb data write-->sync  functions are called, once file is written with 4kb data lfs_fs_size is called to check the file system size.
This sequence works perfectly fine for 16Mb flash but for 64Mb flash, once file size reaches to 12Mb (when sector no 4097, 4098 are used from lfs_ctz_extend) then I get error “../Application/Littlefs/lfs.c:1108:error: Corrupted dir pair at {0x0, 0x1}” from lfs_fs_size.
I’ve modified the relevant configuration settings to use it for 64mb. Please suggest where I’m doing wrong.
// configuration of the filesystem is provided by this struct
const struct lfs_config cfg = {
    // block device operations
    .read  = fs_flash_read,
    .prog  = fs_flash_prog,
    .erase = fs_flash_erase,
    .sync  = fs_flash_sync,
// block device configuration
    .read_size = 256,//16,
    .prog_size = 256,//16,
    .cache_size = 256,//16,
    .lookahead_size = 256,//8,//256,//16,
    .block_count = 16384,//128,
    .block_size = 4096,
    .block_cycles = 1000,//9/30/20: changed for testing 500
};
Thanks,
Anjali

Newfatuser, 2020-11-04:
Hello,
thanks for your inputs, I could resolve the issues and its working now.
one question, can I use littleFS with USB Mass storage device. 
I saw one post but did not get get clear idea from it.
https://github.com/ARMmbed/mbed-os/blob/9e06f07ebe475803b4f3ddd6cd574e5e72293794/usb/device/USBMSD/USBMSD.cpp

please guide.

geky, 2020-11-14:
Ah! Congrats on getting it working, sorry for the late response.

I'm not sure I understand your follow up question.

If you have littlefs on the external flash, and FAT on the USB stick, you can open files on both filesystems and copy over byte-by-byte.

In theory you can put littlefs on the USB stick, but most PCs (Windows/Mac/Linux) don't natively support littlefs. There is [littlefs-fuse](https://github.com/littlefs-project/littlefs-fuse), but you will need a Linux machine, so it likely won't work well for most user-facing applications.

YKMorsy, 2024-06-03:
> Hello, thanks for your inputs, I could resolve the issues and its working now. one question, can I use littleFS with USB Mass storage device. I saw one post but did not get get clear idea from it. https://github.com/ARMmbed/mbed-os/blob/9e06f07ebe475803b4f3ddd6cd574e5e72293794/usb/device/USBMSD/USBMSD.cpp
> 
> I need to read file from USB stick and copy it into external flash, I'll not be copying any file into USB stick and read it on PC. so just one way read operation is possible? please guide.

Hello,
Can you explain what the fix for your problem was? I am running into the same issue.