littlefs-project/littlefs #473: Corrupted file system

dburrell-mio, 2020-10-15:
Hi, I'm using LittleFS on an STM32F7 microcontroller along with a Quad SPI NOR Flash chip from micron all on a custom board. While doing some probing of the various signals between the 2 ICs, there was some file system corruption that happened and as a result, an LFS_ASSERT was tripping when trying to write to one specific file. My question is: what can I do to detect file system corruption before it hits an assert?

geky, 2020-12-08:
Hi @dburrell-mio, thanks for creating an issue. Unfortunately, cleaning up asserts on corrupted filesystems is an area of ongoing improvement.

If you still have the info, which line was the assert on that you hit?

dburrell-mio, 2020-12-08:
Hi @geky I believe it was the assert at line 75 of lfs.c but I've since resolved the issue and didn't document exactly which line it was. It was my code not the LittleFS library that was causing the issue although it was the assert which was causing my system to hang.

geky, 2020-12-18:
Ah, the assert in `lfs_bd_read` did assert often on corrupted filesystems at one point. I believe his was fixed in https://github.com/littlefs-project/littlefs/commit/4677421ababf7be568d27167a51a211749329400 to return LFS_ERR_CORRUPT.