littlefs-project/littlefs #476: ..\littleFS\lfs.c:997:error: Corrupted dir pair at {0x0, 0x1}

eddyv19, 2020-10-25:
I have successfully ported littlefs to a project running on a STM32F103C8T6 (bluepill) with a serial flash (MT25Q128) connected to SPI. 
The basic functions all work. I can create directories, files, write files, read files. Etc. 
Until I reset the processor…  After every reset I get the message:
	
..\littleFS\lfs.c:997:error: Corrupted dir pair at {0x0, 0x1}
Error Mounting Flash Memory: -84
	
I have to reformat the FS to get it to work again.
Is this issue related to my porting or is this a problem with LittleFS? Any help would be much appreciated.!

geky, 2020-11-14:
Hi @eddyv19, thanks for raising an issue. The "corrupted dir pair {0x0, 0x1}" is the first sign that littlefs can't find the filesystem on your device, but I'm not sure what could cause this.

Is it possible you need to call sync or similar on the STM32F103C8T6's flash driver? It's possible this happens if the storage has caches where the writes are lingering, but they aren't being saved persistently. Or if reset causes a flash erase for some reason...

Justdigit, 2020-12-08:
I thought it had to do with a corrupted block. But I received a new ESP32 today and it has the same error in the log.

geky, 2020-12-13:
I still suspect that the block device isn't syncing or flushing properly. Or maybe something during startup corrupts the disk?

littlefs checks that the data-on-disk is what was expected after every write, so we would expect an error to be reported during `lfs_mount` or `lfs_file_close` if the block wasn't able to be written to.

If the block device driver has its own buffer, either on the flash chip or on the MCU, then it needs to be flushed before the data is guaranteed to be there after reboot. littlefs has the `cfg.sync` callback for this.

If that's not the issue then if may be that something else is corrupting the flash during the reboot process.

Justdigit, 2020-12-16:
> I still suspect that the block device isn't syncing or flushing properly. Or maybe something during startup corrupts the disk?
>
> littlefs checks that the data-on-disk is what was expected after every [...]

I have tried two new esp32 boards and different PC's they all give the same log error.

geky, 2020-12-18:
> I have tried two new esp32 boards and different PC's they all give the same log error.

Sorry, to be clear I believe it is a driver issue (in the esp32 HAL, either in the block device driver or the SPI driver). If there is a bug in the driver it would appear on any board.

The first step would be to write a test program that read/writes data to the flash using the same geometry littlefs is configured for, and see if the results are correct even after rebooting.