littlefs-project/littlefs #481: build failed when using the tasking compiler!

snikeguo, 2020-11-03:
IDE:AURIX DS
compiler:CCTC ( tasking company)

```
ctc W562: ["..\littlefs-2.2.1\lfs_util.h" 141/26] unary minus applied to unsigned value
ctc W560: ["../littlefs-2.2.1/lfs.c" 361/24] possible truncation at implicit conversion to type "unsigned char"
ctc W560: ["../littlefs-2.2.1/lfs.c" 738/29] possible truncation at implicit conversion to type "unsigned short int"
ctc W560: ["../littlefs-2.2.1/lfs.c" 980/21] possible truncation at implicit conversion to type "unsigned short int"
ctc W560: ["../littlefs-2.2.1/lfs.c" 1043/16] possible truncation at implicit conversion to type "unsigned char"
ctc W515: ["../littlefs-2.2.1/lfs.c" 1405/200] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 1959/37] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 1981/42] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 1996/39] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 2438/45] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 2778/45] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3206/37] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3331/40] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3350/44] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3440/38] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3642/42] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4055/48] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4089/44] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4138/41] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4201/50] side effects of 'sizeof' operand are ignored
ctc S900: internal consistency check failed - please report
```

geky, 2020-11-14:
Hi @snikeguo, thanks for opening a PR. I'm not sure if I will be able to get to cleaning up lints for a while, but I appreciate having this info.

I can see the reasons for the "unsigned" lints, but I don't know the reason for the "sizeof" lints. They look like they come from LFS_TRACE, is it an issue with printf?

snikeguo, 2020-11-22:
```
ctc W560: ["../littlefs-2.2.1/lfs.c" 361/24] possible truncation at implicit conversion to type "unsigned char"
ctc W560: ["../littlefs-2.2.1/lfs.c" 738/29] possible truncation at implicit conversion to type "unsigned short int"
ctc W560: ["../littlefs-2.2.1/lfs.c" 980/21] possible truncation at implicit conversion to type "unsigned short int"
ctc W560: ["../littlefs-2.2.1/lfs.c" 1043/16] possible truncation at implicit conversion to type "unsigned char"
```
YES !it is an issue with printf

```
ctc W515: ["../littlefs-2.2.1/lfs.c" 1405/200] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 1959/37] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 1981/42] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 1996/39] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 2438/45] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 2778/45] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3206/37] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3331/40] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3350/44] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3440/38] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 3642/42] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4055/48] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4089/44] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4138/41] side effects of 'sizeof' operand are ignored
ctc W515: ["../littlefs-2.2.1/lfs.c" 4201/50] side effects of 'sizeof' operand are ignored
```

 I don't know........sorry...

lucasbrendel, 2024-01-29:
I recently did some investigating and replacing all use of the `LFS_MKATTRS` macro with the actual code by assigning it to a variable and then placing it where the macro was used, removed all of those warnings about sizeof side affects ignored.

Doing that also removes the S900 compiler error about inconsistency checks that prevents compilation from succeeding.

Its a lot of changes to make but for TASKING so far that was needed to be successful.

geky, 2024-02-02:
Oh interesting, it sounds like the warning may be related to how `LFS_MKATTRS` infers the number of attributes?

1. Internally `LFS_MKATTRS` expands such that, roughly:

   ``` c
   lfs_dir_commit(lfs, dir, LFS_MKATTRS({A, B, C}));
   ```

2. Becomes:

   ``` c
   lfs_dir_commit(lfs, dir, (lfs_attr_t[]){A, B, C}, sizeof((lfs_attr_t[]){A, B, C})/sizeof(lfs_attr_t));
   ```

3. And then is evaluated at compile time to, what I assume is more-or-less the same as your rewrite:

   ``` c
   lfs_dir_commit(lfs, dir, (lfs_attr_t[]){A, B, C}, 3);
   ```

CCTC is warning about the second step here because `A, B, C` likely contain side-effects in the sizeof calculation that end up ignored. But this is actually the desired behavior because we don't want to calculate `A, B, C` twice.

You could probably instead change `LFS_MKATTRS` to take the number of attrs as an explicit parameter. That might be easier to maintain.

But changing the attr count to be explicit risks it falling out of date with the actual number of attrs. This is a worse situation than a spurious warning, so I don't think we should change this upstream.

---

To be honest, could you just turn this warning off? I'm struggling to come up with realistic situation where someone would accidentally expect sizeof to evaluate its arguments outside of trivia questions. It seems more harmful than good.

lucasbrendel, 2024-02-03:
I could do an experiment to see, removing the warning. The bigger problem is the last line in the original post

`ctc S900: internal consistency check failed - please report`

The compiler is basically crashing. Making the changes I mentioned gets rid of that crash. I have considered talking to Tasking but given my work environment, it is unlikely I would be able to integrate the patch anytime soon so the modification is the best I got for now.

geky, 2024-02-04:
Oh that is wild. I guess the macro expansions puts more code into the sizeof operator than CCTC can handle? (compiler-internal buffer overflow?)

I guess this is a compiler bug, but glad you found a workaround.

geky, 2024-02-05:
Coincidentally I've been playing around with small fixed-purpose tree editors recently. The following branch should be able to do the LFS_MKATTRS transformation you mentioned above:
https://github.com/geky/qadte/tree/lfs-mkattrs

Though I'll leave it up to you to decide if Rust is more or less difficult than modifying lfs.c by hand.

Eventually, I want to extend this to provide a full transpiler from C99 -> C89 for littlefs. As a side effect this would provide an easy workaround for CCTC. Though this is low-priority at the moment.