littlefs-project/littlefs #483: lfe_file_flush and multiple open file descriptors for the same file caching.

BrianPugh, 2020-11-09:
I am the author of a port of littlefs for the esp32, and am currently trying to resolve this issue:

https://github.com/joltwallet/esp_littlefs/issues/18

problematic code snippet copied here for easy viewing:

```
uint8_t buf1[1] = {'a'};
uint8_t buf2[1] = {};
int fd1 = open("/fs/file.bin", O_RDWR);
assert(fd1 >= 0);
int fd2 = open("/fs/file.bin", O_RDWR);
assert(fd2 >= 0);
lseek(fd1, 0, SEEK_SET);
write(fd1, &buf1, sizeof(buf1));
fsync(fd1);
lseek(fd2, 0, SEEK_SET);
read(fd2, &buf2, sizeof(buf2));
assert(buf1[0] == buf2[0]);  // this fails sometimes.
```

I have narrowed down the issue a little bit. After the `fsync` command, in the line `read(fd2, &buf2, sizeof(buf2));`, the file its pointing at incorrectly reports `file->ctz.size == 0` when it should be `4`. Any idea on whats going on? Thanks!

BrianPugh, 2020-11-09:
I guess what this question really boils down to, is "Can LittleFS handle the same file being opened multiple times concurrently."

embeddedt, 2020-11-11:
Could be related: https://github.com/littlefs-project/littlefs/issues/458

geky, 2020-11-15:
Hello! Interesting project, thanks for putting in the work to port/support it.

Ah yeah, this situation is a bit complicated for littlefs. I've seen questions raised about it's behavior before, but from what I recall it hasn't been a blocker.

In littlefs, when you open the same file multiple times, each `lfs_file_t` gets a COW snapshot of the underlying file. Writing to one open `lfs_file_t` does not affect any of the other open `lfs_file_t`s. You can seek/read/write on other open `lfs_file_t`s as though the file was unmodified. The winner is the last `lfs_file_t` that is sync/closed.

The reason for this behavior is two-fold:

1. We need to worry about memory management. Unlike OS-backed filesystems, we don't have an internal file-descriptor table, so we can't easily deduplicate multiple open `lfs_file_t`s. We do have a linked-list keeping track of open files, so you it's conceivable that we could update all of the other open `lfs_file_t`s on every write. But that would add quite a bit of code size for a less-common use case.

2. Semantically, this matches how the underlying filesystem works. littlefs's files are transactional, so during `lfs_file_sync`, the on-disk state jumps from A to B without intermediary state. This helps write power-resilient programs as you don't have to worry about a half-written file.

   Open files have a different snapshot from the on-disk file in the same way open files have different snapshots from each other.

   This behavior is a bit neutered since `lfs_file_close` implicitly syncs. I've considered adding a `lfs_file_cancel` to allow undoing any amount of file writing, but have been focusing on bug fixes/stability rather than new features.

If you're seeing inconsistent behavior, that definitely is a bug and shouldn't be happening.

I'm still thinking about it, but right now it seems like deduplicating open files may actually make more sense to be implemented in a VFS layer like yours. Open to other thoughts though.

BrianPugh, 2020-11-15:
so far in my vfs layer I've implemented deduplication of `lfs_file_t`, but each file descriptor would need to externally store the current position and mode of the file. Before each operation these values need to be set these flags in the common `lfs_file_t`. Does this sound reasonable, or would you recommend a different method?

geky, 2020-11-15:
Hmm, good point, that does start getting complicated. Another option would be to close+reopen the other `lfs_file_t`s during a sync, however currently that would require keeping track of the full filepath.

Thinking about it more, it looks like your example revolves around `fsync` as the point where the files become consistent. It might not be unreasonable to iterate over open files and force consistency + drop caches during `lfs_file_sync`. It would have the same cost as reopening the files but without needing to store the filepath.

Downside: any changes to other snapshots would be lost. Maybe the current behavior could be chosen by a LFS_O_TRANSACTIONAL flag or similar?

Though I'm still not sure about what the code cost would be.

geky, 2020-11-15:
(A better filesystem would check that caches are in-sync, rather than dropping them, but I think it's safe to say this is one case where we'd rather tradeoff performance for code cost)

geky, 2020-11-20:
Ran into an old related issue, will need to also consider `lfs_file_setattr` behavior: https://github.com/littlefs-project/littlefs/issues/183#issuecomment-515627155

geky, 2020-12-25:
It's a big change-set, so it may take a bit to bring in, but I have pushed up a set of changes that should fix there here: https://github.com/littlefs-project/littlefs/pull/513