littlefs-project/littlefs #496: LFS_ASSERT(pcache->block == LFS_BLOCK_NULL)

9iggor6, 2020-12-04:
Hello, I use littleFS for SD card on STM32H7 with static allocation memory. My configure struct:

```
uint8_t read_buffer_sdio[4096];
uint8_t write_buffer_sdio[4096]; 
uint8_t lookahead_buf_sdio[512];
uint8_t file_buffer[4096];

  obj->lfs.read_buffer = read_buf;
  obj->lfs.prog_buffer = write_buf;
  obj->lfs.lookahead_buffer = lookahead_buf;
  obj->lfs.read_size = 512;
  obj->lfs.prog_size = 512;
  obj->lfs.lookahead_size = 512;
  obj->lfs.cache_size = 4096;
  obj->lfs.context = obj;
  obj->lfs.read = read_sd_card;
  obj->lfs.prog = write_sd_card;
  obj->lfs.erase = erase_sd_card;
  obj->lfs.sync = sync_sd_card;
  obj->lfs.read_size = SD_CARD_BLOCK_SIZE;
  obj->lfs.prog_size = SD_CARD_BLOCK_SIZE;
  obj->lfs.block_size = SD_CARD_BLOCK_SIZE;
  obj->lfs.block_cycles = CNT_CYCLES_ONE_BLOCK;
```
  

 ```
memset(&obj->file_config, 0, sizeof(struct lfs_file_config));
  
  obj->file_config.buffer = file_buffer;  
  obj->file_config.attr_count = 0;
```
 I use function 
 `lfs_file_opencfg(&obj->lfs_config, &obj->lfs_file,  name, LFS_O_WRONLY | LFS_O_CREAT, &obj->file_config);`
for opening file with static buffer.
After first call function:
```
lfs_file_write(&obj->lfs_config, &obj->lfs_file, loggerBuffer_for_write.buffer, 1920);
```
I can assert: 
**LFS_ASSERT(pcache->block == LFS_BLOCK_NULL)**
Dubugging show, that in function **lfs_ctz_extend** give argument **nblock** 

Foe example:
```
In function **lfs_bd_prog**
if (block == pcache->block &&   //block  = 361 pcache->block = 360
                off >= pcache->off &&
                off < pcache->off + lfs->cfg->cache_size)  //Condition is not performed and assert

```
Call stack:
- logger_task
- lfs_file_write
- lfs_ctz_extend
- lfs_bd_prog

When I use cache size 512 bytes, some time the program work correctly.

geky, 2020-12-08:
Hi @9iggor6, thanks for opening an issue. A quick question: What version of littlefs are you using?

cache_size > block_size doesn't actually add anything, littlefs can only cache one block currently. Does your program work when cache_size = block_size (512)?

Maybe there should be an assert in littlefs to protect against this mistake.

9iggor6, 2020-12-08:
Hi geky, LittleFS version 0x00020002. 
You are right, some period of time(~15 min) the programm is working correctly , when cache_size = 512 and then I get assert 
### LFS_ASSERT(pcache->block == LFS_BLOCK_NULL).

geky, 2020-12-13:
Hi @9iggor6, just to clarify, do you hit the assert when cache_size <= 512?