littlefs-project/littlefs #498: Warning about unused lfs_mlist_isopen

dpgeorge, 2020-12-09:
In the latest v2.3.0, there are build warnings about unused `lfs_mlist_isopen()` when asserts are disabled (`-Wunused-function`).  The build that gives this warning is running on GitHub Actions with a macos-11.0 container.

geky, 2020-12-09:
Oh interesting, thanks for reporting this. Out of curiousity, do you know which compiler/version reports this? GCC 7.3.1 doesn't warn since the function is `static inline`, as this is commonly used for header-only functions.

But it's easy enough to cover in a `#ifndef LFS_NO_ASSERT`, `lfs_mlist_isopen` involves a linked-list walk and probably shouldn't be inlined anyways.

dpgeorge, 2020-12-09:
> Out of curiousity, do you know which compiler/version reports this?

No, sorry.  It's probably clang though because it's on Mac.  I can try to find out...

> But it's easy enough to cover in a `#ifndef LFS_NO_ASSERT`

Yes, that's the change I made to get the build to pass without warnings.

geky, 2020-12-09:
Ah that makes sense, currently the only compiler in CI is Clang. I wonder if it does extra checks based on the file extension to avoid warning about header-only functions.

Anyway, here we go: https://github.com/littlefs-project/littlefs/pull/499

dpgeorge, 2020-12-09:
Looks like the compiler that gives the warning is:
```
Apple clang version 12.0.0 (clang-1200.0.32.27)
Target: x86_64-apple-darwin20.1.0
```

Brute4cer, 2021-11-18:
Hi, I am using the latest version of LittleFS 2.4.1 and I get the warning
```
error: 'lfs_mlist_isopen' defined but not used [-Werror=unused-function]
 static bool lfs_mlist_isopen(struct lfs_mlist *head,
             ^
cc1: all warnings being treated as errors
```
I get these errors with the following compiler flags - 
1. LFS_NO_MALLOC
2. LFS_NO_DEBUG
3. LFS_NO_WARN
4. LFS_NO_ERROR
5. LFS_THREADSAFE

when I add the LFS_NO_ASSERT flag the error goes away.

geky, 2022-02-20:
Hmm, @Brute4cer skimming through lfs.c I'm not sure how that could happen.

Is it possible your system's implementation of `assert` does not use its value? Maybe only in release builds?

Maybe we should add `__attribute__((unused))` to this function, though this would be compiler dependent...

lucazader, 2022-05-04:
I can confirm that on my system (stm32h7) the default assert impl will trigger this warning when `NDEBUG` is defined. 
Adding the `LFS_NO_ASSERT` to our cmake build settings for the littlefs lib does limit the warnings.
Adding `__attribute__((used)) ` or `__attribute__((unused))` during compilation will also suppress the warning. with no change on final binary size.

rumbledethumps, 2023-01-10:
Confirming this warning is generated with:
arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
On this Pi Pico project: (Cortex-M0+)
https://github.com/rumbledethumps/rp6502