littlefs-project/littlefs #503: Thoughts: allowing LFS_ASSERT to fail without abort

mon, 2020-12-15:
Had an idea for a PR, but I won't go to the trouble if you think it's not merge-able.

Currently there are two ways to use `LFS_ASSERT`: as a no-op with `LFS_NO_ASSERT`, or as a "kill the entire program" with `assert`. I propose a third option: return `LFS_CORRUPT` and log the error when it occurs, instead of taking down the entire system.

Since LFS_ASSERT is used in functions returning both int and void, I propose adding `...` to LFS_ASSERT, so I can, for example:
`#define LFS_ASSERT(test, ...) if(!(test)) { log_err("%s", #test); return __VA_ARGS__; }`

Alternatively, perhaps splitting the assert into one that returns a value and one that doesn't could work.

geky, 2020-12-18:
My main thought is that it's preferable to find corrupted disk structures at mount/fetch/read time. If, say, `lfs_bd_prog` is called with an out of range block address, that means some logic called `lfs_bd_prog` without checking the address bounds, and that logic should be fixed.

That being said, until we ensure littlefs catches all invalid metadata, this proposal is better than the current state of things.

So I'd be happy to bring in this sort of change, but will also work to replace it in some unknown time frame.

> Alternatively, perhaps splitting the assert into one that returns a value and one that doesn't could work.

I think a second return assert is preferable, it would be better for the control flow to be explicit in case a later rewrite of the function needs to handle cleanup resources.

> log the error when it occurs

There's also the option of logging the assert failure and then trying to continue, though this may just cause more problems than it is worth.

davidefer, 2020-12-22:
Hi
I find the proposal of mon very useful. We have the same problem, an assert in lfs would bring the system down.
We would like very much to perform a sort of graceful degradation in this case. And the error logging is also an issue for us.

mon, 2021-01-20:
With the merging of #514, sorted!

closed 2021-01-20 by mon