littlefs-project/littlefs #506: Use littleFS with small amount of memory.

BorgeatRemy, 2020-12-17:
Hello, 

I want to use littleFS and create a filesystem on a RAM memory. I reserve 4kB of memory in RAM for the filesystem data. I configure block_count at 16 and block_size at 256 bytes in order to have 4 kB available. My configuration of littleFS is below : 
```c
    // block device configuration
    // TODO: tenative size is 256
    .read_size = 16,
    .prog_size = 16,
    .block_size = 256,
    .block_count = 16, /* 4 kbyte */
    .lookahead_size = 8,
    .cache_size = 16,
    .block_cycles = 500,
```
I can mount the filesystem, read file/write file/create directory without problems. 

My goal is to save 100 files of 13 bytes each into this filesystem, but actually, I can't save more than 28-30 files. My files are stored into directory /link/ and their name are 1,2,3,4....

I see there is a *4 file cost storage for small files. So, with this file cost storage, I have 30 * 13 * 4 = 1560 bytes stored.  

So do you have any explanation of what happens here and how I can improve that without increasing too much the filesystem size ? 

Best Regards

geky, 2020-12-18:
Ah, at 16 blocks, you may start running into issues fitting the filesystem tree. Giving the option, LittleFS would rather use more disk space if it can reduce device RAM or improve performance. This makes sense on the systems LittleFS normally runs on, where external flash is ~100x cheaper than device RAM, but may cause issues here.

To disect a bit:
- LittleFS uses 2-blocks for each "metadata pair". These are atomic data structures that store metadata and small files, and is the source of the 4x file cost you found.
- LittleFS uses a linked-list of metadata-pairs for each directory, which means at minimum 1 metadata-pair, or 2 blocks, for each directory. This includes the root directory.
- LittleFS also needs an extra block for wear-leveling, since it needs to copy data from the old block to the new block.

The other issue is that LittleFS needs to store metadata to describe each file, which is at minimum 8 extra bytes + filename, but there's other per-directory metadata such as the revision count, crc, etc, that take up space.

Some thoughts that may improve space usage:
- You can store the files in the root (`/`), that would reduce the number of directories by 1.
- You can set `prog_size` and `read_size` to 1. LittleFS needs to pad the commits otherwise and that padding takes up space.

There's also the option of using LittleFS in "2-block mode". You lose the ability to have directories, but this removes the possible lost space due to LittleFS trying to balance files across the metadata-pairs in a directory, which isn't always perfect:

``` c
    .read_size = 1,
    .prog_size = 1,
    .block_size = 4096/2,
    .block_count = 2,
    .lookahead_size = 8,
    .cache_size = 16,
    .block_cycles = -1, // has no effect with 2 blocks, but wear-leveling does occur in metadata-pairs
```

BorgeatRemy, 2021-01-04:
Thank you for your really nice answer. That's helped me a lot !

closed 2021-01-04 by BorgeatRemy