littlefs-project/littlefs #524: File of size 8 becomes size 0 after repeated writing without unmounting

peschue, 2021-02-03:
I am using littlefs on a flash device with block size 4096 and I have a file with a 64 bit integer (8 byte) as content.

I repeatedly do the following: open (RDWR|CREAT) -> check file size (0 or 8) -> read whole file -> truncate -> rewind -> write whole file -> sync -> close

This works nicely, but not always. Every now and then (around every 100 of the above cycles) file size is reported as 0 and reading also returns only 0 byte, although the previous write returned 8 and sync + close were successful.

I do not remount between these operations.

Is there an inherent limitation about the number of times a file can be opened/rewritten?

peschue, 2021-02-03:
Addition: if I unmount after each writing and then mount again, at some point (last time after 25 times mounting+rewriting+unmounting the file) mount fails with error -22 (invalid parameter).

peschue, 2021-02-03:
If I do not rewind the file but remove it and create it again, the error -22 happens after 3 cycles.

peschue, 2021-02-03:
If I set read_size = 4 and prog_size = 4 this problem disappears.

Now I wonder why, as those are "minimum" read and program sizes.

My chip supports arbitrary reads and writes also across sector boundaries (supposedly). I am using the S25FL256L .

https://www.cypress.com/documentation/datasheets/s25fl256ls25fl128l-256-mbit-32-mbyte128-mbit-16-mbyte-30-v-fl-l-flash

BenBE, 2021-02-03:
Can you try to reproduce this issue with a test program backed by some disk image file (instead of real hardware)?

Does the behaviour change when you change the steps as follows?
1. open(RDWR|CREATE)
2. read(8 bytes)
3. rewind(start of file)
4. write(8 bytes)
5. truncate(at current location, i.e. truncate to 8 bytes)
6. sync()
7. close()

peschue, 2021-02-03:
Yes, indeed, if I truncate to 8 after writing to the file, the problem disappears (but only with read_size=4 and prog_size=4).

Also, while you wrote your message, I edited my previous comment. The problem also disappears if I use read_size=prog_size=4 AND remount after each cycle.

Regarding the test program backend, would bd/lfs_filedb be a reasonable test backend?

BenBE, 2021-02-03:
I think both lfs_filebd and lfs_rambd should work. lsb_filebd has the advantage that you can do snapshots more easily and thus rewind/record operations and their changes for debugging. Ideally you'd end up with a program that just runs 100k cycles of your reproducer and on success finishes with a consistent filesystem.

BenBE, 2021-02-03:
> Yes, indeed, if I truncate to 8 after writing to the file, the problem disappears (but only with read_size=4 and prog_size=4).
> 

This looks quite strongly like a bug. Do things change if you skip the truncation step altogether (shouldn't be necessary in the first place).

> Also, while you wrote your message, I edited my previous comment. The problem also disappears if I use read_size=prog_size=4 AND remount after each cycle.
> 

Okay, another clear sign of some internal state getting corrupted.

peschue, 2021-02-03:
I forked the repo and created a branch. see https://github.com/littlefs-project/littlefs/compare/master...peschue:reproduce-issue-524 

I was not able to reproduce the issue directly, but I got something strange nevertheless after patching my lfs_cfg settings into test.py:

```
$ ./scripts/test.py tests/test_rewrite.toml
====== building ======
built 1 test suites, 1 test cases, 1 permutations
====== testing ======
test_rewrite ✗
====== results ======
tests/test_rewrite.toml:1:failure: test_rewrite#1#1 failed
free(): double free detected in tcache 2

tests passed 0/1 (0.00%)
tests failed 1/1 (100.00%)
```

geky, 2021-02-04:
Thanks for creating an issue and the test case, these really help with debugging issues.

Unfortunately the double free looks like it's caused by an extra `lfs_unmount` inside the for loop, I think you may have meant to comment this out:
[master...peschue:reproduce-issue-524 line 52](https://github.com/littlefs-project/littlefs/compare/master...peschue:reproduce-issue-524#diff-43cdd94275a101a067699368a6afb24553d9e9349e5f0cf796fc440d48d06ee7R52)

> I think both lfs_filebd and lfs_rambd should work. lsb_filebd has the advantage that you can do snapshots more easily and thus rewind/record operations and their changes for debugging.

The reason for lfs_rambd is performance, simulating the filesystem in RAM speeds up CI by ~20x, otherwise we would use filebd for everything since it's easier to debug. The test script defaults to rambd unless you provide `--persist` in which case the disk is written to `tests/test_rewrite.toml.disk`.

This smells strongly like an issue related to caching, which had some changes recently.

geky, 2021-02-04:
This smells like it may be related to recent caching changes, but it's hard to say. It's also possible that caching is hiding failed disk writes either from an fs or driver bug and after a remount the invalid data is revealed.

> Is there an inherent limitation about the number of times a file can be opened/rewritten?

There isn't, but after a metadata block is full LittleFS compacts the block, so it isn't surprising that a bug isn't revealed until after this many writes.

Interestingly that's also about the time when LittleFS would start using blocks other than block 0, since a file size of 8 bytes easily fits in an inlined file.

Have you ran tests on the s25fl to make sure the driver is reading and writing correctly? I messed around with the test case you posted but unfortunately couldn't reproduce that failure. Though it's possible I'm just missing the bug.

peschue, 2021-02-04:
You are right, that was the duplicate unmount. I have a test for my S25FL driver, it is not very sophisticated but it reads repeatedly , erases, programs, and then again reads repeatedly, both at the start and at the end of a block boundary.

peschue, 2021-02-05:
I activated debugging/tracing and printed all prog/read commands and the read content to find out more. Below I show a log with "***" suffixes added by myself. Some reorganization happens while opening the 8-byte file after writing 2 big files and then the 8-byte file has size 0.

Please search for *** and you will find the relevant parts of the log.

```
obc_buffer_id_sequence_obtain: lfsstatus for reading ID is 8, read ID 7
obc_buffer_id_sequence_obtain: next_free_id=7 update_next_free_id=9
thirdparty/littlefs/lfs.c:5244:trace: lfs_file_rewind(0x2000133c, 0x200062dc)

thirdparty/littlefs/lfs.c:5248:trace: lfs_file_rewind -> 0

obc_buffer_id_sequence_obtain: lfsstatus for rewinding NEXTID is 0
thirdparty/littlefs/lfs.c:5212:trace: lfs_file_truncate(0x2000133c, 0x200062dc, 0)

thirdparty/littlefs/lfs.c:5218:trace: lfs_file_truncate -> 0

obc_buffer_id_sequence_obtain: lfsstatus for truncating NEXTID is 0
thirdparty/littlefs/lfs.c:5177:trace: lfs_file_write(0x2000133c, 0x200062dc, 0x200062d0, 8) *** this writes uint64_t "9" ***

thirdparty/littlefs/lfs.c:5183:trace: lfs_file_write -> 8

obc_buffer_id_sequence_obtain: lfsstatus for writing updated_next_free_id 9 is 8
thirdparty/littlefs/lfs.c:5126:trace: lfs_file_close(0x2000133c, 0x200062dc)

littlefs read block 0 offset 220 buffer 0x2000617c size 4
littlefs read ff ff ff ff
littlefs prog block 0 offset 200 buffer 0x2000c010 size 16
littlefs prog block 0 offset 216 buffer 0x2000c010 size 4
littlefs read block 0 offset 200 buffer 0x2000bff8 size 16
littlefs read 70 1f e0 0c 09 00 00 00 00 00 00 00 70 1f e0 0c *** here the value of 0x 09 00 00 00 00 00 00 00 is visible
littlefs read block 0 offset 216 buffer 0x2000bff8 size 4
littlefs read d6 43 f8 ff
thirdparty/littlefs/lfs.c:5131:trace: lfs_file_close -> 0

obc_buffer_id_sequence_obtain: lfsstatus for close is 0
littlefs_unlock
obc_buffer_id_sequence_obtain: exit (first_id_output = 7)
obc_buffer_store_adler_signal_packet: entry (0x20006368)
littlefs took semaphore
littlefs_mount: exit with 0x2000133c
obc_buffer_store_adler_signal_packet: filename 7.pkt type 0
thirdparty/littlefs/lfs.c:5090:trace: lfs_file_open(0x2000133c, 0x200062e4, "7.pkt", b02)

littlefs read block 1 offset 0 buffer 0x200061b8 size 4
littlefs read f9 01 00 00
littlefs read block 0 offset 0 buffer 0x200061bc size 4
littlefs read fa 01 00 00
littlefs read block 0 offset 4 buffer 0x2000bff8 size 16
littlefs read f0 0f ff f7 6c 69 74 74 6c 65 66 73 2f e0 00 10
littlefs read block 0 offset 20 buffer 0x2000bff8 size 16
littlefs read 00 00 02 00 00 10 00 00 f0 1f 00 00 20 00 00 00
littlefs read block 0 offset 36 buffer 0x2000bff8 size 16
littlefs read ff ff ff 7f fe 03 00 00 20 00 1c 1e 4e 45 58 54
littlefs read block 0 offset 52 buffer 0x2000bff8 size 16
littlefs read 49 44 00 00 18 03 31 2e 70 6b 74 20 30 00 0d 9f
littlefs read block 0 offset 48 buffer 0x20006130 size 4
littlefs read 4e 45 58 54
littlefs read block 0 offset 68 buffer 0x2000bff8 size 16
littlefs read 1a 00 00 60 00 00 00 20 30 0c 0d 32 2e 70 6b 74
littlefs read block 0 offset 84 buffer 0x2000bff8 size 16
littlefs read 20 30 00 0d a0 1a 00 00 28 01 00 00 20 30 04 0d
littlefs read block 0 offset 100 buffer 0x2000bff8 size 16
littlefs read 33 2e 70 6b 74 20 30 00 0d a1 1a 00 00 60 00 00
littlefs read block 0 offset 116 buffer 0x2000bff8 size 16
littlefs read 00 20 30 1c 0d 34 2e 70 6b 74 20 30 00 0d a2 1a
littlefs read block 0 offset 132 buffer 0x2000bff8 size 16
littlefs read 00 00 28 01 00 00 00 30 0c 00 07 00 00 00 00 00
littlefs read block 0 offset 148 buffer 0x2000bff8 size 16
littlefs read 00 00 20 00 08 0d 35 2e 70 6b 74 20 30 00 0d a3
littlefs read block 0 offset 164 buffer 0x2000bff8 size 16
[... 840 lines cut ...]
thirdparty/littlefs/lfs.c:5248:trace: lfs_file_rewind -> 0

obc_buffer_id_sequence_obtain: lfsstatus for rewinding NEXTID is 0
thirdparty/littlefs/lfs.c:5212:trace: lfs_file_truncate(0x2000133c, 0x200062dc, 0)

thirdparty/littlefs/lfs.c:5218:trace: lfs_file_truncate -> 0

obc_buffer_id_sequence_obtain: lfsstatus for truncating NEXTID is 0
thirdparty/littlefs/lfs.c:5177:trace: lfs_file_write(0x2000133c, 0x200062dc, 0x200062d0, 8)

thirdparty/littlefs/lfs.c:5183:trace: lfs_file_write -> 8

obc_buffer_id_sequence_obtain: lfsstatus for writing updated_next_free_id 3 is 8
thirdparty/littlefs/lfs.c:5126:trace: lfs_file_close(0x2000133c, 0x200062dc)

littlefs read block 1 offset 1144 buffer 0x2000617c size 4
littlefs read ff ff ff ff
littlefs prog block 1 offset 1124 buffer 0x2000c010 size 16
littlefs prog block 1 offset 1140 buffer 0x2000c010 size 4
littlefs read block 1 offset 1124 buffer 0x2000bff8 size 16
littlefs read 70 1f dc 0e 03 00 00 00 00 00 00 00 70 1f dc 0c
littlefs read block 1 offset 1140 buffer 0x2000bff8 size 4
littlefs read 9a b2 43 87
thirdparty/littlefs/lfs.c:5131:trace: lfs_file_close -> 0
```

peschue, 2021-03-04:
I was able to track this down to an issue in the flash driver, where writing 12 byte sometimes wrote a 13th byte.

I found the issue by writing json description of all read/write/erase operations and the data and analyzing with a python script.

Closing this issue. Thank you for your support!

closed 2021-03-04 by peschue