littlefs-project/littlefs #527: Spec Understanding ? | Could Delete result in immediate compaction?

thrasher8390, 2021-02-16:
When reading through the spec the answer I get to is "Yes", Delete would cause a compaction because it updates the manifest with the file id with the tag `LFS_TYPE_DELETE` which could put the metadata pair over the 50% limit and hence trigger compaction.

BUTTT I would be very happy if I was wrong because it would be nice if Deleting files was very fast. E.g. if I want to reset a device I'd rather the reset or factory default be nearly immediate instead of taking upwards of seconds due to some throttling I've invoked on the spi communication. Hence, I've been looking to see if LFS does anything fancy where it assumes the file id's tag is a bitmap and clears a bit which sets it up for future deletion. E.g. I delete now and any future file creation or modification would result in compaction and deletion of the file id.

TLDR; Will delete result in immediate compaction, yes or no?

geky, 2021-02-16:
Unfortunately deletes may result in a compaction.

LittleFS stores metadata in an append-only log, and doesn't rewrite data on storage without an erase first (there are some flash chips that don't support bit nanding). So to perform a delete, LittleFS actually writes a new commit that indicates the file is deleted. This commit could exceed the available space in the metadata block, triggering a compaction. And then during compaction the file would be cleaned up completely.

The 50% limit is actually when a metadata block gets split into two metadata blocks. This is to avoid an exponential growth since near-full metadata blocks could end up trigger compactions every commit.

---

You can't avoid appending a new commit, but you can combine more metadata in a single commit (which also reduces padding used to align each commit to the program size). One way to do this with the filesystem API is to replace files with `lfs_rename`. This is atomic and writes the metadata for the delete and new file in a single commit. It might still trigger a compaction, but it at least should trigger fewer compactions.

thrasher8390, 2021-02-17:
Thanks @geky, that is the answer I expected. I also appreciate the additional information with the `lfs_rename`, greatly appreciated.

closed 2021-02-17 by thrasher8390