littlefs-project/littlefs #533: lfs_error .. No more free space

baranovus, 2021-02-25:
Hello.
Sorry for asking the question, which might been already answered. I have an error message  **lfs_error .. No more free space**. I saw in discussions that it is necessary to increase "erase size to up to 6 blocks...". Could somebody explain what is it? 
Does it mean that I have to set .**block_size** equal to 6 flash erasable block size values? For example, if my flash allows erasing 4096 byte blocks, should I set block_size = 6*4096?

geky, 2021-03-23:
Ah, I'm not sure the original source, but "memory allocation" should really be "storage". LittleFS requires ~6 blocks of storage to work with. There's not really a lower bound on memory, though a large cache_size = less bus transactions = faster.

But 6 blocks of storage is the bare minimum. If you have multiple or large files you may need more storage. How much data are you trying to write?

baranovus, 2021-03-23:
Hello and thanks!
I need to keep about 10 files. Most of them are about 100 bytes and  a couple of them may be about 100 K.
I think I am confused with the word "block". When we say "block" do we mean the size of erasable block of the flash memory (4096 Bytes in my case)?  If yes, do I need ~24 K of storage and that will be the size of my cache? 
Now I have:
#define CACHE_SIZE          256 
#define LOOK_AHEAD_SIZE     64
#define READ_SIZE           CACHE_SIZE
#define PROG_SIZE           CACHE_SIZE
#define ERASABLE_BLOCK_SIZE 4096

.read_size = READ_SIZE,
.prog_size = PROG_SIZE,
.block_size = ERASABLE_BLOCK_SIZE,

Where am I wrong?

jimarcher, 2021-03-29:
@baranovus any luck with this?

baranovus, 2021-03-30:
Hello Jim.
After some modifications which seem to be unconnected with the problem the error disappeared,  but of course I am not satisfied because I don't understand what had been done wrong.

jimarcher, 2021-04-02:
Okay thanks very much for your reply.