littlefs-project/littlefs #536: Using LittleFS for Winbond W25Q16JW

JonJust, 2021-03-03:
Hello,

I am currently working on implementing LittleFS on the Winbond W25Q16JW through SPI. I have verified that I am able to interface with the chip manually, being able to read and write to it.

The W25Q16JW array is organized into 8192programmable pages of 256-bytes each. Up to 256 bytes can be programmed at a time. Pages can be erased in groups of 16 (4KB sector erase), groups of 128 (32KB block erase), groups of 256 (64KB block erase) or the entire chip (chip erase). The W25Q16JWhas 512 erasable sectors and 32 erasable blocks respectively. 

After tuning my config to fit the specs of the chip, I am having issues being able to format the chip. After commenting out the LFS_DEBUG printf functions from lfs.c (as they were causing hard faults), lfs_format() is returning an error code of -28 in main. 

Is there anything I am doing explicitly wrong? Thanks!

Here is what my code look like:

Declaring structs:

```
// variables used by the filesystem
lfs_t lfs;
lfs_file_t file;
struct lfs_config cfg;

uint8_t lfs_read_buf[256];
uint8_t lfs_prog_buf[256];
uint8_t lfs_lookahead_buf[128];	// 128/8=16
uint8_t lfs_file_buf[256];
```

Linking I/O funtions:

```
int block_device_read(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, void *buffer, lfs_size_t size)
{
	w25->readData((uint32_t)(block*c->block_size + off), (char*)buffer, size);
	return 0;
}

int block_device_prog(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, const void *buffer, lfs_size_t size)
{
	w25->pageWrite((uint32_t)(block*c->block_size + off), (char*)buffer, size);

	return 0;
}

int block_device_erase(const struct lfs_config *c, lfs_block_t block)
{
	w25->sectorErase(block*c->block_size);
	return 0;
}

int block_device_sync(const struct lfs_config *c)
{
	return 0;
}
```

Config function:
```

// configuration of the filesystem is provided by this struct
void LFS_Config(void)
{
	// block device operations
	cfg.read  = block_device_read;
	cfg.prog  = block_device_prog;
	cfg.erase = block_device_erase;
	cfg.sync  = block_device_sync;

	// block device configuration
	cfg.read_size = 256;
	cfg.prog_size = 256;
	cfg.block_size = 4096;
	cfg.block_count = 4096;
	cfg.cache_size = 512;
	cfg.lookahead_size = 128;
	cfg.block_cycles = 500;

	cfg.read_buffer = lfs_read_buf;
	cfg.prog_buffer = lfs_prog_buf;
	cfg.lookahead_buffer = lfs_lookahead_buf;
//	cfg.file_buffer = lfs_file_buf;
}
```

I/O I am having issues with in main.cpp

```
  int res = 0;
  LFS_Config();
  res = lfs_format(&lfs, &cfg);
  res = lfs_mount(&lfs, &cfg);
  res = lfs_file_open(&lfs, &file, "", LFS_O_CREAT | LFS_O_RDWR);
  res = lfs_file_write(&lfs, &file, "abcedfghijklmn", 14);
  res = lfs_file_read(&lfs, &file, readBuffer, 14);
```

Functions that explicitly handle the i/o on the chip:

```
void SPIFlash::readData(uint32_t addr, char* data, uint32_t len) {
	//Reads a length of data from a given address
	  cs_set();
	  outBuffer[0] = W25_READ_DATA;
	  outBuffer[1] = (addr >> 16) & 0xFF;
	  outBuffer[2] = (addr >> 8) & 0xFF;
	  outBuffer[3] = addr & 0xFF;
	  sendSPI(outBuffer, 4);
	  recvSPI(data, len);
	  cs_reset();
	  usDelay(1);
}

void SPIFlash::pageWrite(uint32_t addr, char* data, uint32_t len) {
	/* [Public Method]
	 *  Writes to a given page in memory.
	 *
	 *  len defaults to the page size. If len is less than the page size,
	 *  the function will only write to a page however much data is specified.
	 */

	if (len > PAGE_SIZE) len = PAGE_SIZE;

	enableWrite(true);
	cs_set();
	outBuffer[0] = W25_PAGE_PROGRAM;
	outBuffer[1] = (addr >> 16) & 0xFF;
	outBuffer[2] = (addr >> 8) & 0xFF;
	outBuffer[3] = addr & 0xFF;
	for (int i = 0; i < len; i++) outBuffer[i+4] = data[i];
	sendSPI(outBuffer, len + 4);
	cs_reset();
	enableWrite(false);
	usDelay(1);
}

void SPIFlash::sectorErase(uint32_t addr) {
	/* [Public Method]
	 * Wipes a given sector from memory
	 */
	  cs_set();
	  outBuffer[0] = W25_SECTOR_ERASE;
	  outBuffer[1] = (addr >> 16) & 0xFF;
	  outBuffer[2] = (addr >> 8) & 0xFF;
	  outBuffer[3] = addr & 0xFF;
	  sendSPI(outBuffer, 4);
	  cs_reset();
}
```

gtaska, 2021-03-03:
Here's something I encountered with a similar Winbond device (a *JV variant). Given that you say that you can already read & write to the device this may not be the problem, but I thought it was worth checking, none-the-less.

You don't show what enableWrite() does. This could either be playing with the Write Protect (WP) line, and/or sending a WriteEnable SPI instruction. The fact that you don't have the chip select line asserted before enableWrite() makes me think that you might only be toggling the WP line. (And you don't seem to be doing anything in sectorErase() to enable writes)

You need to make sure that both the WP line is deasserted, *as well as* sending a WriteEnable (06h) SPI instruction before **every** PageProgram, or Sector/Block/Chip erase SPI transaction.

I got caught out on this, and thought it was only required once per page program. Because I have to split my page programs up into multiple SPI transactions, only the first partial page program operation would succeed.

JonJust, 2021-03-03:
Hi! I appreciate the response but I don't think thats the issue. I have two macros in use to handle the CS pin, and my enableWrite() sends the command. Did you ever get a file system working on the similar chip? Thanks!

```
#define cs_set() HAL_GPIO_WritePin(WIN_FLASH_CS_PIN,GPIO_PIN_RESET)
#define cs_reset() HAL_GPIO_WritePin(WIN_FLASH_CS_PIN,GPIO_PIN_SET) // NEED TO CHANGE DEPENDING ON LAYOUT OF BOARD
```

```
void SPIFlash::enableWrite(bool state) {
	/* [Public Method]
	 * Sends a sequence to the chip to enable
	 * writing. This MUST be done right before
	 * attempting to write to the chip, else
	 * the chip won't write.
	 */
	if (state) outBuffer[0] = W25_WRITE_ENABLE;
	else outBuffer[0] = W25_Write_DISABLE;
	cs_set();
	sendSPI(outBuffer, 1);
	cs_reset();
}
```

gtaska, 2021-03-04:
Yes, I got it working on a W25Q32JV (16J/32J is the memory density, W/V is the operating voltage).

Unless I am missing something, you don't appear to be enabling writes when you erase.

Also, in your configuration you have:

>	cfg.block_size = 4096;
>	cfg.block_count = 4096;

This would be 4096 * 4k = 16MBytes, however your part is 16Mbit.

BTW, according to the datasheet, the Write Disable (04h) is not needed after the operation:

> Note that the WEL bit is automatically reset after Power-up and upon completion of the Write Status Register, Erase/Program Security Registers, Page Program, Quad Page Program, Sector Erase, Block Erase, Chip Erase and Reset instructions.

JonJust, 2021-03-04:
I notice in retrospect that the documentation refers to as sectors as 4096 bytes and blocks as 32Kbytes in size for the erase feature. I am treating the 4Kbyte sectors as blocks when implementing into lfs. That shouldn't be a problem, should it?

gtaska, 2021-03-04:
No, that shouldn't cause a problem - you just need to make sure you use the correct erase operation - SectorErase (20h) or BlockErase (52h for 32kB block, D8h for 64kB block).

Your approach for determining when a write/erase operation has completed probably isn't the best. Rather than using a fixed delay, you should read the status register and check for the ERASE/WRITE IN PROGRESS flag. A page program has a typical write time of 0.8ms and a max time of 3ms. A 1us delay isn't going to be much use. (And you don't seem to check if/when erases complete)

Finally, I don't know how the device will respond if you assert the write protect line before a write/erase has completed.

JonJust, 2021-03-04:
I implemented a function to wait for the register to clear as well as de-asserting the CS line after the write is complete, and the issue persisted.... To simplify I swapped it out for a broad 5ms delay between writes to no dice.... Are you at liberty to share what your read/write/erase functions look like? 

Thank you so much for your assistance

gtaska, 2021-03-05:
Unfortunately, I am not in a position to share the company's work product, however have you considered looking at using other existing W25qxx drivers? For example [here](https://github.com/nimaltd/w25qxx), a driver for STM devices?

Having a known-good implementation for the W25Qxx device would then narrow down the problem space to either the littlefs configuration, or an implementation issue.

At the very least, it would be good to contrast & compare your driver.

JonJust, 2021-03-05:
Gave this a shot, got the library implemented and working but still getting a -28....

```
int block_device_read(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, void *buffer, lfs_size_t size)
{
	W25qxx_ReadPage((uint8_t*)buffer, block*c->block_size, off, size);
	return 0;
}

int block_device_prog(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, const void *buffer, lfs_size_t size)
{
	W25qxx_WritePage((uint8_t*)buffer, block*c->block_size, off, size);
	return 0;
}

int block_device_erase(const struct lfs_config *c, lfs_block_t block)
{
	W25qxx_EraseSector(block*c->block_size);
	return 0;
}

// configuration of the filesystem is provided by this struct
void LFS_Config(void)
{
	// block device operations
	cfg.read  = block_device_read;
	cfg.prog  = block_device_prog;
	cfg.erase = block_device_erase;
	cfg.sync  = block_device_sync;

	// block device configuration
	cfg.read_size = 256;
	cfg.prog_size = 256;
//	cfg.block_size = 32678;
//	cfg.block_count = 128;
	cfg.block_size = 4096;
	cfg.block_count = 512;
	cfg.cache_size = 512;
	cfg.lookahead_size = 128;
	cfg.block_cycles = 500;

	cfg.read_buffer = lfs_read_buf;
	cfg.prog_buffer = lfs_prog_buf;
	cfg.lookahead_buffer = lfs_lookahead_buf;
//	cfg.file_buffer = lfs_file_buf;
}
```

JonJust, 2021-03-05:
Update: I figured out what was causing the formatting issues! The size of my read and program buffers were not equal to the cache_size value I provided in the initialization. I am going to close this for now, hopefully this will be helpful to someone.

closed 2021-03-05 by JonJust

gtaska, 2021-03-07:
Great to hear. My next step was to point out that you are dropping any data over a page-length in size in the write function. If your write function can automatically split a longer write (or read for that matter) up, then you don't need to have the constraint that page size == cache size.

simon88, 2024-10-23:
Hi @JonJust @gtaska 
I have the same issue with the same flash. @JonJust do you remember you config `lfs struct` for W25Q16 flash ?

here is what I've tried

````
struct lfs_config cfg_lfs = {
        .read = lfs_read,
        .prog = lfs_prog,
        .erase = lfs_erase,
        .sync = lfs_sync,

        .read_size = 256,
        .prog_size = 256,
        .block_size = 4096,
        .block_count = 512,
        .cache_size = 256,
        .lookahead_size = 128,
        .block_cycles = 500,

        .read_buffer = read_buffer,
        .prog_buffer = prog_buffer,
        .lookahead_buffer = lookahead_buffer
};
````

Thanks a lot

JonJust, 2024-10-26:
`
#define BUF_SZ 256
#define SECTOR_SIZE 4096
#define PAGE_SIZE 256
#define SECTOR_COUNT 512

// configuration of the filesystem is provided by this struct
void lfs_configure(void) {
	// block device operations
	// cfg is a global variable held in main.cpp.
	// It would probably be better to set this up without global variables, but
	// hey if it aint broke don't fix it.

	cfg.read  = block_device_read;
	cfg.prog  = block_device_prog;
	cfg.erase = block_device_erase;
	cfg.sync  = block_device_sync;

	cfg.read_size = BUF_SZ;
	cfg.prog_size = BUF_SZ; // read and prog need to be identical
	cfg.block_size = SECTOR_SIZE;
	cfg.block_count = 512; // Found in the documentation of the flash chip
	cfg.cache_size = BUF_SZ; // Need to be identical size to read and prog
	cfg.lookahead_size = 256;
	cfg.block_cycles = 1500; // Number of cycles one can write to a block before lfs engages wear leveling.

	cfg.read_buffer = lfs_read_buf;
	cfg.prog_buffer = lfs_prog_buf;
	cfg.lookahead_buffer = lfs_lookahead_buf; // Link the buffers

	cfg.name_max = 64;
	cfg.file_max = LFS_FILE_MAX;
	cfg.attr_max = LFS_ATTR_MAX;
	cfg.metadata_max = SECTOR_SIZE;

}`

My Init Function looked something like this