littlefs-project/littlefs #537: Many small files

Happybolt, 2021-03-04:
cfg = lfs.LFSConfig(block_size=4096, block_count=64, prog_size=16, read_size=16, cache_size=512, lookahead_size = 1024, block_cycles = 500)

I write in a loop each time a new file with static dimensions (500 bytes), in the root directory. And I get that I can write 75 files , paylaod = 75 * (500 + 32) (32 - len name file)  = 39900 (bytes), on Flash witch size block_size * block_count = 262144, that is, the efficiency of using memory resources is  15,22% This is normal? Doesn't look very good.
 I did a similar test on MCU, the result is the same.. 
 For comparison, I took SPIFFS,under the same conditions obtained efficiency ~70% , yes littleFS much faster, that's cool! But right now I want to store many files 100 - 1000 bytes.  
 Perhaps I am doing something wrong?

geky, 2021-03-05:
Hi @Happybolt, I want to emphasize that I think SPIFFS is a great filesystem. This whole embedded-filesystem space is a really interesting area of tradeoffs, and SPIFFS does a great job at utilizing the space on disk. As I think you noticed, LittleFS trades of disk utilization for performance.

I don't think you're doing anything wrong, I think this is just a case where LittleFS doesn't fit as well as SPIFFS.

The main feature you are leaning on is LittleFS's "inline files", which is where LittleFS stores files directly in metadata blocks if the files fit in the cache size (500 byte files < 512 byte cache so this should be happening). But the way LittleFS stores metadata/inline files incurs a roughly 4x cost in storage. This, along with extra metadata for managing directories and wear leveling are likely what pushes the storage cost so high.

SPIFFS will probably always store data more compactly than LittleFS, and I think SPIFFS is a better choice for really tiny filesystems built in internal flash or small eeprom chips. LittleFS has an interesting overlap in that it can run on these small pieces of storage, but the main value of this is for use cases where you want to reduce your application to a single filesystem to deduplicate code cost, if say you have a larger filesystem in off-chip flash with LittleFS on it already.

If you need to store data compactly in a small amount of storage, I think SPIFFS is the way to go.

Happybolt, 2021-03-05:
Thanks for the answer, but SPIFFS slow especially if i have 16MB.  Opening a file takes a tremendous amount of time (if files already exist at the beginning of memory). To avoid this, you will have to use your own caching for each file, and this MAX_LENGHT_PATH + spiffs_pix = ~36 bytes on per file, for a large number of files, cache storage becomes cumbersome. And on litlleFS  it seems to work fine on large media,

After reading the documentation again, I decided to try :) 
lfs_file_open(_stat->_handlerFs, &file, _path, LFS_O_RDWR | LFS_O_CREAT | LFS_F_INLINE);
I added a flag LFS_F_INLINE , and the result has grown a lot, it makes me happy. But I noticed that now closing the file takes an incredibly long time. I can't understand what the addiction is, is it possible to somehow get around it?

Happybolt, 2021-03-05:
And there were still problems with creating a folder (not quite on the topic, but maybe there is a quick answer) 
lfs_mkdir(&statFs[1]._handlerFs, "myFolder/"); return LFS_ERR_CORRUPT   
I try "myFolder/" and "myFolder", the result is unchanged ( 
My sequence (maybe this is the problem):
  err = lfs_mount(&lfs, &cfg);
  if(err)
  {
      lfs_format(&lfs, &cfg);
      lfs_mount(&lfs, &cfg);
	  __NOP();
  }
  lfs_mkdir(&statFs[1]._handlerFs, "myFolder/");
  It is mounted normally, I can write (only if in the root folder)

geky, 2021-03-05:
Hmm, you should not be passing LFS_F_INLINE yourself, those flags (LFS_F*) are only used internally (should probably be moved into lfs.c) and will likely break things.

LittleFS should automatically inline the file if it's possible, the fact that you were able to fit more files could mean either:
- LittleFS is being to conservative on when it inlines files
- Passing LFS_F_INLINE causes LittleFS's internal logic to break and drop files

It would be interesting to see if you get better usage with 2x the block size (4096) since there is a check in place that the file fits in 1/4 the block since otherwise this can cause the metadata-compaction algorithm to never terminate.

To check against the second possibility, you could try writing some repeating byte pattern to the file and manually check if the byte pattern for each file shows up on the disk. You could do this locally with a [test script](https://github.com/littlefs-project/littlefs/blob/master/tests/test_files.toml#L2) and test.py's --persist flag.

---

I suspect passing LFS_F_INLINE is what's causing lfs_mkdir to break. Your code looks correct, but if LFS_F_INLINE goes wrong it would mess up the metadata that is also used to store directories.