littlefs-project/littlefs #539: lsf_file_open returns -36 (LFS_ERR_NAMETOOLONG) When trying to create file

JonJust, 2021-03-05:
Hello,

I am currently working on implementing LittleFS on the Winbond W25Q16JW through SPI.

I have managed to get my file system to be able to format the drive and mount to it.... However whenever I go to create a file lfs_file_open returns a -36. According to the enum in lfs.h, this would imply that the name of the file I am trying to create is too long, however this makes little sense as name_max defaults to 255... 

Here is what my code to interface with lfs looks like:

```
  int res = 0;
  LFS_Config();
  res = lfs_mount(&lfs, &cfg);

  if (res != 0) { // If unable to mount chip, wipe, format, and remount
	  W25qxx_EraseChip();
	  res = lfs_format(&lfs, &cfg); // ONLY NEED TO MOUNT ONCE
	  res = lfs_mount(&lfs, &cfg);
  }

  res = lfs_file_open(&lfs, &file, "ts", LFS_O_CREAT | LFS_O_RDWR);
  res = lfs_file_write(&lfs, &file, "abcedfghijklmn", 14);
  res = lfs_file_read(&lfs, &file, readBuffer, 14);
```

Here are my configurations: 

```
// variables used by the filesystem
lfs_t lfs;
lfs_file_t file;
struct lfs_config cfg;

uint8_t lfs_read_buf[256];
uint8_t lfs_prog_buf[256];
uint8_t lfs_lookahead_buf[128];	// 128/8=16
uint8_t lfs_file_buf[256];

int block_device_read(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, void *buffer, lfs_size_t size)
{
	W25qxx_ReadPage((uint8_t*)buffer, block*c->block_size, off, size);
	return 0;
}

int block_device_prog(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, const void *buffer, lfs_size_t size)
{
	W25qxx_WritePage((uint8_t*)buffer, block*c->block_size, off, size);
	return 0;
}

int block_device_erase(const struct lfs_config *c, lfs_block_t block)
{
	W25qxx_EraseSector(block*c->block_size);
	return 0;
}

int block_device_sync(const struct lfs_config *c)
{
	return 0;
}

// configuration of the filesystem is provided by this struct
void LFS_Config(void)
{
	// block device operations
	cfg.read  = block_device_read;
	cfg.prog  = block_device_prog;
	cfg.erase = block_device_erase;
	cfg.sync  = block_device_sync;

	// block device configuration
	cfg.read_size = 256;
	cfg.prog_size = 256;
	cfg.block_size = 4096;
	cfg.block_count = 512;
	cfg.cache_size = 256;
	cfg.lookahead_size = 128;
	cfg.block_cycles = 500;

	cfg.read_buffer = lfs_read_buf;
	cfg.prog_buffer = lfs_prog_buf;
	cfg.lookahead_buffer = lfs_lookahead_buf;

	cfg.name_max = LFS_NAME_MAX;
	cfg.file_max = LFS_FILE_MAX;
	cfg.attr_max = LFS_ATTR_MAX;
}

```

gtaska, 2021-03-08:
Not sure if this is the problem, but...

Be careful with the lfs vs driver definitions of page, sector, block. They are different.

I believe that the W25qxx_Read/WritePage functions are supposed to accept an offset & size no more than the page size (0-255), however you are passing in a lfs block offset & size (0 - 4095).

You need to figure out which page in the block you are addressing.

```
pageInBlock = offset / PAGE_SIZE
offsetInPage = offset % PAGE_SIZE
pageAddress = block * BLOCK_SIZE + pageInBlock * PAGE_SIZE
W25qxx_WritePage((uint8_t*)buffer, pageAddress, offsetInPage, size)
```

Furthermore, if the `size` is larger than the page, or causes the write to straddle a page boundary, then you'll need to issue separate page write commands for each page. (And implementing this logic should also allow you to then use a larger cache size, which you currently avoid)

JonJust, 2021-03-08:
Ahhhh that makes a lot more sense! I am now able to successfully create and open files, and verify that they are in the file system. The system doesn't throw off any errors when I go to write, returning the number of bytes that it was told to write, but is having trouble reading the files into buffers..... I think there is a chance that I am "straddling" a page address somewhere, but can verify that the size of the write never exceeds 256... I have some tinkering to do, but will update the thread when I solve the issue, or make a new one if I can't.

Thank you so much for all the help you have provided me. This is my first time implementing a file system on bare metal and your advice was instrumental in allowing me to make the progress I did.

closed 2021-05-06 by JonJust