littlefs-project/littlefs #542: file list

WanCooder, 2021-03-16:
How to read the file list？

geky, 2021-03-23:
You can read the files with `lfs_dir_read`, it behaves like [readdir](https://linux.die.net/man/3/readdir).

``` c
lfs_dir_t dir;
struct lfs_info info;
int err = lfs_dir_open(&lfs, &dir, "/");
if (err) {
    return err;
}

while (true) {
    int res = lfs_dir_read(&lfs, &dir, &info);
    if (res < 0) {
        return err;
    }
    
    if (!res) {
        break;
    }
    
    printf("%s %d", info.name, info.type);
}

err = lfs_dir_close(&lfs, &dir);
if (err) {
    return err;
}
```

alexandrehagihara, 2022-04-14:
`
    int res = lfs_dir_read(&lfs, &dir, &info);
    if (res < 0) {
        return err;
    }
`
Can I let the directory opened here?

geky, 2022-04-15:
Hi @alexandrehagihara, good catch. We should make sure to release any resources if there is an error.

I've updated the example in the comment above.

Not closing the directory would have only been acceptable if the device halted afterwards.