littlefs-project/littlefs #558: Add the freeBytes() function to the lib

ullix, 2021-05-16:
I am switching in use between LittleFS and FFat and one little thing makes it cumbersome and the code is cluttered with #if and #elif in file system related functions:

The FFat lib has the functions:
```
//    size_t totalBytes();
//    size_t freeBytes();
```

while LittleFS (and SPIFFS) has:
```
//    size_t totalBytes();
//    size_t usedBytes();
```

If you could add  `freeBytes();` it would be easier.

I did modify the lib myself - even I could handle that successfully - but this limits portability of my code, and needs to be repeated and not forgotten at the next upgrade.

geky, 2021-06-11:
Wouldn't it be better to add usedBytes to FFat then?

But since freeBytes can be defined as `totalBytes() - usedBytes()`, I don't think it's worth adding/supporting freeBytes to the API. Besides don't you need ifdefs to choose between `lfs_fs_size()` and the SPIFFS/FFat equivalents? Or is this behind an RTOS/SDK API?

ullix, 2021-06-11:
Sure, it does not matter which of the libs gets the missing function. Preferably both gets the addition.

I just hope it does not become a Ping-Pong of suggestions that the "other lib" does it! 

It is a rather trivial thing to do. But it avoids cluttering the code with a lot of ifdefs.

geky, 2021-06-12:
But won't you need an ifdef anyways for `lfs_fs_size` vs `f_getfree`?