littlefs-project/littlefs #559: Size file not change when append data

KhanhPham0102, 2021-05-21:
I'm using NAND Flash W25N01GV to read and write data with config below:
```
read_size        = 1023, 
prog_size        = 1023,
block_size       = W25N_MAX_MAIN_ARRAY_SIZE * W25N_MAX_PAGES_PER_BLOCK, // 131072
block_count      = W25N_MAX_BLOCK_NUMBER, // 64
cache_size       = 2048,
lookahead_size   = W25N_MAX_BLOCK_NUMBER * 2, // 128
block_cycles     = 500
```

Code test:
```
uint8_t data_buff[] = "aaaaaaaaaaaaaaaaaaaaaaaaaaaaa";
lfs_open("abc", O_RDWR | O_CREAT);
lfs_write(data_buff, strlen(data_buff));
lfs_sync();
lfs_close();

uint8_t abc[] = "abcbcbcbcbcbcbxc";
lfs_open("abc", O_WRONLY | O_APPEND);
lfs_write(abc, strlen(abc));
lfs_sync();
lfs_close();

uint8_t pop[] = "qqqqqqqqqqqqqqqqqqqqqqqq";
lfs_open("abc", O_WRONLY | O_APPEND);
lfs_write(pop, strlen(pop));
lfs_sync();
lfs_close();
```
I write data and read the file size and position of the pointer write, then after writing the first and second package successfully, when opening the file to write 3rd package, the file size returns to the size when saving the first package. 

Show Log:
[image]

Debug using read_buffer array:
Write 1st:
[image]

Write 2nd:
[image]

Write 3rd:
[image]

I have tried changing the read cursor position as well as changing the file size but the result is still the same. What is my current problem here ?

e107steved, 2021-05-25:
May well not be the problem, but why are you using a read and program size of 1023? A power of two, such as 1024, is "normal" here, and may well be a littlefs requirement.

KhanhPham0102, 2021-05-27:
> May well not be the problem, but why are you using a read and program size of 1023? A power of two, such as 1024, is "normal" here, and may well be a littlefs requirement.

KhanhPham0102, 2021-05-28:
> May well not be the problem, but why are you using a read and program size of 1023? A power of two, such as 1024, is "normal" here, and may well be a littlefs requirement.

I know that, normally these numbers would be (2^n) but when I set for read_size and program_size to 1024 I get below error:
[image]

When I set (2^n - 1) the error will not appear. I set up read and write function like below for littlefs:

`static int read_littlefs_wrapper(
  const struct lfs_config * c, lfs_block_t block, lfs_off_t off, void * buffer, lfs_size_t size)
{
  (void)c;

  return flash_read(block, off / 1024, off % 1024, (uint8_t *)buffer, size); // block_num, page_num, column_num
}
static int write_littlefs_wrapper(const struct lfs_config * c, lfs_block_t block, lfs_off_t off,
  const void * buffer, lfs_size_t size)
{
  (void)c;

  return flash_write(block, off / 1024, off % 1024, (uint8_t *)buffer, size);
}
`