littlefs-project/littlefs #560: LittleFS with a database?

epistax, 2021-05-24:
We are successfully using LittleFS off of a small iMX for storage of some finite data generated at runtime. We will soon be writing out log data as well. This poses an added challenge because it'll likely be small piece of information that accumulate indefinitely. Of course we'll need to do something about a data limit and potentially round-robining by pruning older entries. I'm looking for suggestions in terms of a mechanism that sits on top of Littlefs, or perhaps a technique for directly using Littlefs that makes the best use of our storage available.

sqlite is too large for our application--code space is limited. 
We could certainly just write files.. 1.rec, 2.rec, 3.rec, 4.recv, each with  maybe 16 bytes, but that seems wasteful.

I can't say the exact requirements because I don't know them yet. I imagine speed of data recovery will not be critical. Maximizing the amount of data that will fit within a particular space while also still allowing for load balancing will be important. Anything we come up with will be tested using a virtual flash device to check the wear over an accelerated lifetime so we have a check / safety net in place.

Any suggestions?

Thanks!

e107steved, 2021-05-25:
As a non-expert in the field, I would expect a "proper" database to be rather slow. To me, "database" implies the existence of an index file in some form, which is going to have to be randomly updated every time you add a record. LittleFS is not good at that.
Rather than write a single record to each file (which is, as you say, going to be somewhat wasteful of space, and may cause you to hit other limits), why not use multiple size-limited files; start a new file when you hit a size limit on the current one.

epistax, 2021-05-25:
If the append functionality is considered efficient then this is something to look into. Thanks