littlefs-project/littlefs #568: Comments on lfs_config struct fields do not describe bits vs. bytes

Jak-o-Shadows, 2021-06-14:
The comments for the various size parameters in the _lfs_config_ struct do not describe whether they are in bytes vs. bits. This would help newcomers.

geky, 2021-06-22:
Good point, will elaborate

eclig, 2021-07-23:
What now?  Are they bytes or bits?

geky, 2021-08-03:
Ah yes! My bad, I did not answer the actual question. I realize I still need to update this in the codebase but in the meantime:

All units are bytes.

---

Internally all units are bytes as well, it's fairly rare for the filesystem to operate on bits. One exception is the lookahead buffer, which is a bitmap containing one bit per block. Humorously it is the only field that mentions the unit in the documentation, because it is also configured in bytes. This was chosen for consistency.

closed 2022-03-22 by geky