littlefs-project/littlefs #570: File System corrupting after writing a file a certain amount of times

JonJust, 2021-06-15:
I have been working on getting LittleFS working on the STM32 Microcontroller with the Winbond w25qxx over SPI. I have managed to get the file system working for the most part, however I am having issues with the system corrupting after writing a few files to the system. The amount of files required to corrupt the system seems to deviate and not be related to file size; I have been able to write 4-5 100kb+ files before corruption and seen corruption from the 3rd 2kb file. I have also been able to identify that the issue persists when deleting the same file and rewriting it. 

The issue happens after I try to close a file. I will write to the target file and get a return value indicating a successful write, however trying to close the file returns a -84 (LFS_ERR_CORRUPT). Once this happens, the entire file system is corrupt. I wrote a testing function that writes Smash Mouth's All Star to the chip 10 times. In the function I split up the writes into chunks of 284 because I end up needing to do that in the final product.

```
void lfs_debug() {
	  msDelay(500); // Ensures this method won't be called a bunch of times which could kill the chip

	  // Trying to isolate issue where writing files multile times using RR3 causes file system to corrupt...
	  // We will need to take a ~2kb text file and write it over and over again until the issue appears.

	  // https://tomeko.net/online_tools/cpp_text_escape.php?lang=en << Useful website to convert large amounts of text into a c++ string
	  char textBuffer[4096] = <THIS STRING CONTAINS THE LYRICS OF SMASH MOUTH'S ALL STAR (~2kb)>

	  char* textPtr; // We will be splitting up writes into chunk sizes to emulate the issue so we want to start by pointing the start of the string
	  const unsigned int textLen = strlen(textBuffer); // Get the size of the file
	  unsigned int remaining = textLen;
#define CHUNK_SIZE_MAX 284
	  unsigned int chunkSize;

	  char readBuffer[4096]; // This is the buffer we will be reading into
	  char dirBuffer[1024]; // we will read all directory information into here

	  int res = lfs_sprint_dir(dirBuffer,"");

	  memset(readBuffer, 0, 1024); // Clear buffer

	  res = serDebug->sendString("\aDirectory at boot: \r\n");
	  res = serDebug->sendString(dirBuffer);
	  res = serDebug->sendCRLF(2);

	  char fn[64] = "FFFFFFF1.TXT";

	  bool filesMatch = true;

	  for (int i = 0; i < 10; i++) {
		  while(serDebug->isWriting()) msDelay(10);

		  remaining = textLen; // Set variables before going through loop
		  textPtr = textBuffer;
		  filesMatch = true;

		  res = lfs_remove(&lfs, fn);

		  res = lfs_file_open(&lfs, &file, fn, LFS_O_CREAT | LFS_O_RDWR); // Creates the file
		  //while(serDebug->isWriting()) msDelay(20); // Wait for transmission to end

		  if (res < 0) { // If the file cannot be opened, send an error
			//while(serDebug->isWriting()); // Wait for transmission to end
			res = serDebug->sendString("\aError: Couldn't create new file at index: ");
			res = serDebug->sendNumberToAsciiString(i);
			res = serDebug->sendCRLF(2);
			continue; // bypass the rest of the loop as the file couldn't be opened
		  }

		  while (remaining > 0) { // Break file up into chunks of 284 bytes and write piece by piece
			  if (remaining > CHUNK_SIZE_MAX) chunkSize = CHUNK_SIZE_MAX;
			  else chunkSize = remaining;
			  res = lfs_file_write(&lfs, &file, textPtr, chunkSize);

			  if (res < chunkSize) { // Write failed, send an error
				//while(serDebug->isWriting()); // Wait for transmission to end
				res = serDebug->sendString("\aError: Write didn't work!");
				res = serDebug->sendCRLF(2);
				res = lfs_file_close(&lfs, &file); // Saves the file
				continue; // bypass the rest of the loop as the file couldn't be opened
			  }

			  textPtr += chunkSize;
			  remaining -= chunkSize;
		  }
		  res = lfs_file_close(&lfs, &file); // Saves the file

		  if (res < 0) { // Couldn't close file, send an error!
			//while(serDebug->isWriting()); // Wait for transmission to end
			res = serDebug->sendString("\aError: Couldn't close file after write!");
			res = serDebug->sendCRLF(2);
			res = lfs_remove(&lfs, fn);
			continue; // bypass the rest of the loop as the file couldn't be opened
		  }

		  res = lfs_sprint_dir(dirBuffer,""); // output directory
		  res = serDebug->sendString("\aDirectory at write index: ");
		  res = serDebug->sendNumberToAsciiString(i);
		  res = serDebug->sendCRLF(2);
		  res = serDebug->sendString(dirBuffer);
		  res = serDebug->sendCRLF(1);

		  res = lfs_file_open(&lfs, &file, fn, LFS_O_CREAT | LFS_O_RDWR); // Opens the file to be read

		  if (res < 0) { // If the file cannot be opened, send an error
			//while(serDebug->isWriting()); // Wait for transmission to end
			res = serDebug->sendString("\aError: Couldn't open file after writing!");
			res = serDebug->sendCRLF(2);
			continue; // bypass the rest of the loop as the file couldn't be opened
		  }

		  res = lfs_file_read(&lfs,&file,readBuffer,textLen);

		  for (unsigned int j = 0; j < textLen; j++) if (readBuffer[j] != textBuffer[j]) filesMatch = false; // Check equiviliance of buffers

		  if (filesMatch) serDebug->sendString("\aFile verified!");

		  res = lfs_file_close(&lfs, &file); // Saves the file
		  res = serDebug->sendCRLF(2);
	  }
}
```

This is the output in RealTerm:
```

Board Initialized...         
                                                                                
Wiping w25...                        
                                                                                
Directory at boot:                       
dir    0B .                                                
dir    0B ..
                         
Directory at write index: 0                        
dir    0B .                     
dir    0B ..                           
fil   2KB FFFFFFF1.TXT
                                        
File verified!
                                                              
Directory at write index: 1
dir    0B .                           
dir    0B ..  
fil   2KB FFFFFFF1.TXT
                                                         
File verified!
                                                             
Directory at write index: 2                                                     
dir    0B .                      
dir    0B ..
fil   2KB FFFFFFF1.TXT
                                                        
File verified!
                                         
Directory at write index: 3                                                       
dir    0B .                                       
dir    0B ..                   
fil   2KB FFFFFFF1.TXT
                                                                    
File verified!
                                                                      
Directory at write index: 4                                                                                                    
dir    0B .                                          
dir    0B ..                                
fil   2KB FFFFFFF1.TXT

File verified!
                                                          
Error: Couldn't close file after write!
                                                                       
Error: Couldn't create new file at index: 6
                                                           
Error: Couldn't create new file at index: 7
                                
Error: Couldn't create new file at index: 8
                    
Error: Couldn't create new file at index: 9

  

```

JonJust, 2021-06-15:
THINGS TRIED:
	- Adding flags in writeBinaryFile() to turn LED1 on permenantly in the event that res becomes an error... The LED never turned on
	- Switching to another board, implying the w25 is not at fault
	- Increased the lookaheadbuf size from 256 to 512 in the LFS config
	- Reduced the lookaheadbuf size from 256 to 128 to 16
	- Increased cfg.block_cycles from 500 to 1500
	- Reducing max file name size from 255 to 64
	- Reduced max file size to 262144
	- Tried increasing read/write/cache buffer sizes from 256 to 512. (broke entire system)
	- Tried disabling wear-leveling all together by setting block_cycles to -1
	- Tried explicitly setting cfg.metadata_max to SECTOR_SIZE (4096)
	- Unmounting and remounting the file system each time before doing a re-write

JonJust, 2021-06-17:
UPDATE: The issue seems to be related to lfs_remove..... When I don't remove the file I don't get a corruption.... Strangely enough I cannot seem to see the files by doing a directory read; only the first file I wrote can be seen... Regardless, they DO seem to exist in the system, as I am able to read the files

JonJust, 2021-06-17:
UPDATE2: I fixed the issue with files not showing up  in the directory, but attempting to write allstar8.txt corrupts allstar7.txt, and I become unable to open any new files after. Attempting to open a file freezes the entire program.

Directory at write index: 7

dir    0B .                                                               
dir    0B ..                                                              
fil   2KB allstar0.txt                                                        
fil   2KB allstar1.txt                                                       
fil   2KB allstar2.txt                                                    
fil   2KB allstar3.txt                                                       
fil   2KB allstar4.txt                                                       
fil   2KB allstar5.txt                                                 
fil   2KB allstar6.txt        
fil   2KB allstar7.txt

File verified!

Directory at write index: 8

dir    0B .                                                               
dir    0B ..                                                              
fil   2KB allstar0.txt                                                        
fil   2KB allstar1.txt                                                       
fil   2KB allstar2.txt                                                    
fil   2KB allstar3.txt                                                       
fil   2KB allstar4.txt                                                       
fil   2KB allstar5.txt                                                 
fil   2KB allstar6.txt        
fil   0B allstar7.txt

File corrupt!

JonJust, 2021-06-18:
Update3: I solved the issue! I had a flaw in my disk I/O where I was performing the index to address calculation twice, which was causing the system to accidentally work in some regards.

closed 2021-06-18 by JonJust

drpioneer, 2021-08-10:
> Update3: I solved the issue! I had a flaw in my disk I/O where I was performing the index to address calculation twice, which was causing the system to accidentally work in some regards.

Hello.
Show me where in your code the problem was?
What has changed in the code to fix the problem?
Thanks in advance.

JonJust, 2021-08-10:
```
int block_device_prog(const struct lfs_config *c, lfs_block_t block,
	lfs_off_t off, const void *buffer, lfs_size_t size) {

	uint32_t pageInBlock = off / PAGE_SIZE;
	uint32_t offsetInPage = off % PAGE_SIZE;
	uint32_t pageAddress = (block*SECTOR_SIZE) + (pageInBlock*PAGE_SIZE);

 	w25->pageWrite(pageAddress, offsetInPage, (char*)buffer, size);
	return 0;
}
```

This is what your disc i/o SHOULD look like. Keep in mind that pageAdress is an INDEX rather than an actual address, I just haven't bothered to update the variable name. Hope that helps!

JonJust, 2021-08-10:
Specifically the issue I had in this post was that my erase function wasn't working properly; I'd check to make sure that's working for you

drpioneer, 2021-08-16:
Thanks to your information, it became clear that the problem was located in the functions that provide interaction between LittleFS and the flash memory driver. The fault was found and eliminated. Thank you.